prusa3d/PrusaSlicer #3416: Unable to compile 2.2.0-alpha2 on OSX

spuder, 2019-12-28:
### Version

version_2.2.0-alpha2

220e2dbfec58cbbcff548d8211a17f0dcbe32aab

### Operating system type + version
10.14.6

I have succesfully compiled requisites

```
mkdir build
cd build
cmake ..
make
```

When I attempt to build PrusaSlicer, cmake is unable to find wxWidgets 3.1 (however it *does* find my local install of 3.0)

```
CMake Error at /usr/local/Cellar/cmake/3.16.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find wxWidgets: Found unsuitable version "3.0.4", but required is
  at least "3.1" (found -L/usr/local/lib;;;-framework IOKit;-framework
  Carbon;-framework Cocoa;-framework AudioToolbox;-framework
  System;-framework
  OpenGL;-lwx_osx_cocoau_html-3.0;-lwx_osx_cocoau_adv-3.0;-lwx_osx_cocoau_gl-3.0;-lwx_osx_cocoau_core-3.0;-lwx_baseu-3.0)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.16.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:391 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.16.2/share/cmake/Modules/FindwxWidgets.cmake:1008 (find_package_handle_standard_args)
  src/CMakeLists.txt:53 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/spencerowen/Code/github/PrusaSlicer/build/CMakeFiles/CMakeOutput.log".
```

It seems that wxWidgets is packages as `wxmac` in the brew repository. 3.0.x is the latest available version
If I uninstall wxmac and everything that depends on it, I get error that it can't find wxWidgets
```
brew uninstall elixir
brew uninstall erlang
brew uninstall wxmac
```

```
CMake Error at /usr/local/Cellar/cmake/3.16.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS) (Required is at least version "3.1")
Call Stack (most recent call first):
```

Shouldn't cmake be finding the compiled in deps/build, not my OS version of wxWidgets?

spuder, 2019-12-28:
Others have had so many issues with wxWidgets

https://github.com/prusa3d/PrusaSlicer/issues/2077
https://forum.prusaprinters.org/forum/english-forum-general-discussion-announcements-and-releases/help-porting-slicer-wxwidgets-environment-variables/

lukasmatena, 2019-12-29:
How exactly do you invoke cmake?

spuder, 2019-12-30:
Following these steps

https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md

I've build the dependencies

```
cd deps
mkdir build
cd build
cmake ..
make
```

Then I attempt to build PrusaSlicer

```
cd ../../   #Go up a directory to ensure in root directory of project
mkdir build
cd build
cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"
```

I've also tried just `cmake ../` 

I'm pretty sure I"m in the correct directory 

```
-- IGL NOT found, using bundled version...
CMake Error at /usr/local/Cellar/cmake/3.16.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS) (Required is at least version "3.1")
```

spuder, 2019-12-30:
After being able to reproduce this half a dozen times, I cleaned the build directory and tried again and it worked 

# 🤷‍♂

closed 2019-12-30 by spuder

lukasmatena, 2021-12-09:
@spuder Got back here from #7412 to provide more information. If purging the build directory helped, it was quite likely caused by CMake, which caches certain variables and states between invocations. So in some cases one incorrect call to cmake can poison the build directory so the subsequent cmake calls don't do what is expected. As far as I know, this is a danger inherent to using CMake. So if you first tried to link system libraries and only then set the `CMAKE_PREFIX_PATH` to your compiled dependencies, it might explain the problem. How exactly the caching works is far beyond me and it possibly differs across CMake versions. This does not look like a problem in PrusaSlicer itself.

In the latter case (switching generator to XCode) CMake refuses to comply if it sees that previous invocation used something else.