prusa3d/PrusaSlicer #3418: [Feature Request] Bridge Direction Enforcers

NovaViper, 2019-12-29:
## Operating system type + version
Windows 10 1909 64 bit

### 3D printer brand / version + firmware version (if known)
Creality Ender 3

### Behavior
This is related to #3414. It seems the bridging algorithm treats bridges going in two different directions that originate from the same point as one bridge. In one example, whenever I print out [this test cube](https://www.thingiverse.com/thing:30171), it creates the properly long string bridging on two sides, but oddly, it goes in the wrong direction for the other two directions. An demonstration of this is shown here: 
[image]

I had another instance where I was printing a phone case that had stripes in it, only 4 of the sides did an up and down motion while the 5th stripe did a straight line along the direction of the strip, which caused it to fail a little bit. This is how the slicer processed the bridging for this:
[image]

This issue seems to be present at least a year (I tested all the way to 1.30), the bridging logic seems to go nuts whenever a model introduces multiple bridging directions that come from the same point. While setting a custom bridging angle does fix the problem with the phone case since it only needs to bridge in that one direction; however the cube is still wrong because the algorithm is treating the bridges as 1 bridge because they all come in the same points.

Hence, this 'bug' may be some sort of limit of what the algorithm can detect. Therefore, I am requesting the ability to define bridging direction using enforcers that way you could run multiple bridging directions in close proximity

liftbag, 2019-12-29:
Even an automatic bridges management algorithm would be welcome. Now I can't remember what results, but other slicers already have automatic bridge direction management.

NovaViper, 2026-09-02:
This issue still there even in v2.9.6