prusa3d/PrusaSlicer #3419: [Feature Request] Option to dump slicer parameters in JSON as input for Post Processing script

mkudzia84, 2019-12-29:
### Rationale:
Currently the only way to pass Slicer parameters onto the post processing script is by adding them to the gcore file by adding them in the Start G-Code section.

Example:
`;e3dtc-set:filament0_temperature={ temperature[0] }
;e3dtc-set:filament1_temperature={ temperature[1] }
;e3dtc-set:filament2_temperature={ temperature[2] }
;e3dtc-set:filament3_temperature={ temperature[3] }
;e3dtc-set:filament0_first_layer_temperature={ first_layer_temperature[0] }
;e3dtc-set:filament1_first_layer_temperature={ first_layer_temperature[1] }
;e3dtc-set:filament2_first_layer_temperature={ first_layer_temperature[2] }
;e3dtc-set:filament3_first_layer_temperature={ first_layer_temperature[3] }
;e3dtc-set:filament_temperature_standby_delta={ -standby_temperature_delta }`

This makes the slicer add the specified parameters in the gcode file - at which point the post processing script has to parse the whole gcode file and read them.
This approach works - but with number of parameters increasing - and with a very limited UI interface for setting the custom G-Codes  injections within Prusa Slicer - it doesn't scale well and is very clumbersome.

### Solution:
Provide a dump of all print slicer parameters in a temporary JSON file that is created in a work folder, prior to execution of post processing script.
The JSON dump file then gets deleted after the post processing script is executed.

Dump could be toggled on/off by extra checkbox next to the section specifing post-processing CMD line.
The JSON file name and location could be configured in that section as well.

This would provide an easy query mechanism for post processing script to check the settings of gcore file.

bubnikv, 2020-01-02:
> Currently the only way to pass Slicer parameters onto the post processing script is by adding them to the gcore file by adding them in the Start G-Code section.

The parameters are also added to scipt's environment.

bubnikv, 2020-03-15:
Can you use the evironment?