prusa3d/PrusaSlicer #3421: Zmizí filamenty

joka-cz, 2019-12-29:
### Version
_2.2.0 Alpha 2 win64_

### Operating system type + version
_Win 10 Pro 64 bit CZ_

1. když spustím program, nabídka filamentů - OK. - viz chyba1.3mf
2. nastavím 3 barvy jedním etxtruderem - OK - viz chyba2.3mf
3. když uložím nastavení tiskárny, zůstane v nabídce pouze právě zvolený filament - CHYBA - chyba3.3mf
4. když změním Nastavení tisku (výšku a kvalitu vrstvy), zmizí všechny filamenty - CHYBA - chyba4.3mf
5. když se pokusím přidat/odebrat filament, program na to nereaguje - výbrané filamenty se nezobrazí v nabídce

Josef Karpíšek
[mizí_filamenty.zip](https://github.com/prusa3d/PrusaSlicer/files/4008321/mizi_filamenty.zip)

bubnikv, 2020-03-15:
bude opraveno v PrusaSlicer 2.2.0-rc4, bohužel je ale možné, že ve Vašem případě je už Vaše konfigurace rozbitá, takže pokud 2.2.0-rc4 nepomůže, smažte prosím konfigurační adresář.

closed 2020-03-15 by bubnikv

joka-cz, 2020-03-15:
OK, díky za odpověď