prusa3d/PrusaSlicer #3422: Model hit when recovering from a crash on sparse wipe tower

rocky00717, 2019-12-29:
Describe the bug
When using the new Sparse wipe tower feature, a crash occurred on the wipe tower about 80% into the print. I had the model in the bottom left and the tower on the top right. When the machine went to recover from the crash, it went to x0, y0 and maintained the wipe tower height thus knocking the model off the bed.

To Reproduce
Steps to reproduce the behavior:

wait until printed model is higher than wipe tower
Force a crash when wiping
Observe....
3D Model and Slic3r Configuration Export
[Yoshi.zip](https://github.com/prusa3d/PrusaSlicer/files/4008365/Yoshi.zip)

Expected behavior
The print head should move up to the model height before recovering from crash to avoid hitting the model.

Desktop (please complete the following information):

OS: Windows 10 version 1903
PrusaSlicer Version: 2.2.0-alpha2+win64
Additional context
An easy workaround is to move the model to the bottom right and the wipe tower to the top left.

bubnikv, 2020-01-02:
> An easy workaround is to move the model to the bottom right and the wipe tower to the top left.

or to disable the crash detection

bubnikv, 2020-03-15:
Thanks for heads up. We will add a note to the manual for you to disable crash detection with sparse wipe tower.

closed 2020-03-15 by bubnikv