prusa3d/PrusaSlicer #3423: Scars on outter walls after printing the inside

StefNiem, 2019-12-29:
### Version
PrusaSlicer  2.2.9

### Operating system
Windows 10 Pro Version 1903

### 3D printer brand / version + firmware version
Original Prusa i3 MK3 (Upgraded to MK3S), Firmware FIRMWARE 3.8.1 (For MK3S)

### Behavior
The print has an ugly scar on the outter wall .. its always at the same height where there is a bottom/top layer next to it on the inside (for example on a box or smth like that).
The same mistake doesnt occur on S3D or Cura (all printed with equal settings).
Extruder is calibrated aswell as the extrusion multiplier.
Prints in the pictures are 0,3mm layerheight.

_Is this a new feature request?_
No (i guess)

#### Project File (.3MF) where problem occurs
[Print.zip](https://github.com/prusa3d/PrusaSlicer/files/4008366/Print.zip)

1st white image show the issue, 2nd one is printed with S3D, orange one is printed with prusaslicer but its just another model)

[image]
[image]
[image]

ProfDrYoMan, 2019-12-29:
I see the same often.

Neolker, 2019-12-29:
More evidence here https://m.facebook.com/groups/675831176090951?view=permalink&id=922016038139129

ProfDrYoMan, 2019-12-29:
Closed group?

photog0411, 2020-01-01:
Basically you won't get this issue with S3D, and will only get this issue slightly with Cura, but you will get this issue a lot with Slic3r or Prusa Slicer. If it doesn't happen in S3D it's something that can definitely be fixed but who knows how hard it is to be fixed in Slic3r. No one ever wants to address this issue! 

Maybe @supermerill  can fix it in his fork?

photog0411, 2020-01-01:
It always happens, on every print, whenever the in-fill becomes solid inside, like a shelf in a box. No matter how many perimeters are set, or how slow the print is made, the issue still persists in multiple prints. No settings change it. It is a problem with the Slic3r algorithim pushing the exterior perimeters for those layers farther out than the rest of the external perimeters.

[image]
[image]

photog0411, 2020-01-01:
**Related Issues:**
https://github.com/prusa3d/PrusaSlicer/issues/2070
https://github.com/prusa3d/PrusaSlicer/issues/1719
https://github.com/prusa3d/PrusaSlicer/issues/991

photog0411, 2020-01-01:
@StefNiem 

Can you slice the same object with Slic3r++ and run a print, let me know if the scars on the side disappear? 
https://github.com/supermerill/Slic3r/releases/tag/2.2.46.0
Be sure the settings "enforce 100% fill volume" is checked on perimeters settings

DanioZg, 2021-07-05:
Hello,
Any updates on that issue? I got exactly same problem every print. Walls are thicker when there are someting filled inside.
Slicer: Prusa Slicer 2.3.1
Printer: Prusa Mini+

supermerill, 2021-07-05:
It's often due to a huge difference in the printing time. As so, the filament didn't retract the same way.
You can try with more perimeters, slow perimeters and fast solid infill.

closed 2026-09-29 by prusaslicer-bot[bot]