prusa3d/PrusaSlicer #3424: 2.2.0 alpha2 interferes with wireless USB mouse

pgregory, 2019-12-29:
### Version

2.2.0-alpha2+win64-201912221340

### Operating system type + version
Windows 10  version 1809

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro

### Behavior

The 2.2 alpha version seems to interfere with the operation of the wireless mouse I have. It's a cheap one, Daffodil, but simply based on a USB dongle, works fine for everything else, including PrusaSlicer 2.1.1. The mouse seems to work ok for the first 30-60 seconds after starting PrusaSlicer 2.2.0 alpha2, but then the mouse stops working, the touchpad continues to operate normally, suggesting perhaps PrusaSlicer is doing something on the USB bus? Once this happens, the only way to get the mouse back is to remove the USB dongle, and then plug it back in. If PrusaSlicer is still running, this will reinstate normal mouse operation for 30-60 seconds again, before it is lost in the same way. If PrusaSlicer is no longer running, everything returns to normal.

bubnikv, 2020-01-02:
@enricoturri1966 implemented support for the 3D mouse into PrusaSlicer 2.2.0-alpha1. You are certainly first person reporting such an issue. It may be a driver issue on your side. Let's @enricoturri1966 look into this issue.

pgregory, 2020-01-02:
Just to be clear, it's not a "3D mouse", just an ordinary 2D wireless mouse. Just in case by "3D mouse" you mean something like a spacemouse, it's not. 

In addition, I realised I actually had the printer (Ender 3 Pro) connected by USB at the time, and printing from Cura, so I'll re-test and see if it was just a weird interaction between the various elements. I have no experience with the code, but I would suspect from a laymans perspective, it's something to do with some part of the code polling USB looking for devices, be those 3D mouses or printers.

bubnikv, 2020-03-15:
Please retest with the upcoming PrusaSlicer 2.2.0-rc4. We have switched from polling the USB devices to Win32 events.

There is still polling of removable drives every 2 seconds in PrusaSlicer 2.2.0-rc4, though this is what Cura does for years, so I don't expect that to be causing issues. Enumerating USB devices every 2 seconds likely caused the issues you described though.

Closing. Please reopen if the issue is not resolved.

closed 2020-03-15 by bubnikv