prusa3d/PrusaSlicer #3425: Incorrect raft z-distance [unified issue tracker]

nordurljosahvida, 2019-12-30:
This aims to be a single thread to track all raft z-distance related issues, all of which don't seem to be gaining enough attention individually, nor are the suggestions given in any way conclusive.

Also, I haven't seen some conclusive research onto the topic that provides data as to what actually currently happens under different scenarios.

### Version

<= 2.2.0-alpha2, testing in this post with 2.1 stable

### Operating system type + version

several if not all

### 3D printer brand / version + firmware version (if known)

seemingly irrelevant

### Behavior

The distance / air gap that is left between the raft and the first model layer is vastly incorrect, leading the object in several cases to be printed in mid-air, far above the last layer of the raft.

### Previous threads mentioning the problem

- https://github.com/prusa3d/PrusaSlicer/issues/2823
- https://github.com/prusa3d/PrusaSlicer/issues/2766
- https://github.com/prusa3d/PrusaSlicer/issues/118
- https://github.com/prusa3d/PrusaSlicer/issues/102
- https://github.com/prusa3d/PrusaSlicer/issues/1445 [different but related]

#### Project File (.3MF) where problem occurs

### Analysis

First off, there *needs* to be clarity and documentation about what we're talking about.

I'm using MK3S printer [standard 0.4 nozzle] with generic PLA and 0.15 quality profile. 4 layers of raft and 0.1mm contact z distance. supports are enabled but don't seem to affect this, but are helpful to understand the thickness of the first model layer by visual comparison.

1. In prints with raft enabled, does the `first layer height` value apply to the first *raft* layer or to the first *model* layer, or both? **Answer: first raft layer only**

2. In my tests the first model layer above the raft is printed at 0.4mm layer height independently of the  [with a 0.4mm nozzle by the way!], respecting the `contact z distance` setting, so after the last raft layer the toolhead moves 0.5mm up and prints a 0.4mm layer 0.1mm vertically off the raft. The setting seems to be: `first model layer above raft height = nozzle diameter`. This doesn't work!

3. Raft layer height [except first raft layer] is not customizable by the user, in my tests it's printed at 0.3mm layer height with 0.4 nozzle and 0.45mm layer height with 0.6 nozzle. The setting seems to be: `non-first raft layer height = nozzle diameter *3/4`. This is reasonable.

3. If you set `first layer height` to `0.4` in the above 3mf prusaslicer file:

* First layer / first raft layer: **0.4mm**
* Second layer / second raft layer: **0.3mm**
* Third layer / third raft layer: **0.3mm**
* Fourth layer / fourth raft layer: **0.3mm** [toolhead height 1.30mm]
* Fifth layer / first model layer: **0.4mm** with **0.1mm** z-airgap [toolhead height 1.80mm]

If instead you set `first layer height` to `0.2mm`:

* First layer / first raft layer: **0.2mm**
* Second layer / second raft layer: **0.3mm**
* Third layer / third raft layer: **0.3mm**
* Fourth layer / fourth raft layer: **0.3mm** [toolhead height 1.10mm]
* Fifth layer / first model layer: **0.4mm** with **0.1mm** z-airgap [toolhead height 1.60mm]

### Expected behaviour:
1] The first model layer printed at 1:1 nozzle size layer height is in my opinion way too high, and produces in almost all cases a random print in the middle of the air.
2] The first model layer should be customizable, if not at least set to a reasonable fraction of the nozzle diameter.
3] The `first layer height` setting **must** specify that it's being applied to first `raft` layer and not first `model` layer above the raft in prints with a raft.

Shama71, 2019-12-30:
I think that need to make separate settings for the raft so that there are no misunderstandings and add more parameters, such as width, etc.

sungdzda, 2020-01-01:
This issue deserves more attention. It's surprising to me that most users are not even alarmed with the raft/support quality of Prusa printers. One negative effect is the fact that Prusa printers are uniquely characterized by the "reverse elephant foot" phenomena which significantly impacts functionality of the printed model. Also see #2767. It is worth noting that the raft/support implementation of Slic3r/PrusaSlicer differs grossly from industry customs and standards. Furthermore, Prusa development seems to be driven by a few unhealthy opinions (promoting true raftless printing style and the use of polyesters which never has any warp issue, while by the large ignoring other use cases).

nordurljosahvida, 2020-01-02:
I really do not understand the statements on slic3r's website where they say that rafts were for pre-heated bed / pre-mesh bed levelling days. I print with PETG every day and not only do I print on the PEI sheet with PETGs rather than the textured one [except for small objects], it still warps off! So rafts are definitely still really important IMO, let's make them work.

bubnikv, 2020-01-02:
> I print with PETG every day and not only do I print on the PEI sheet with
PETGs rather than the textured one [except for small objects], it still
warps off! So rafts are definitely still really important IMO, let's make
them work.

I don't understand how raft can help with warping off. If there is an
excessive stress, something somewhere has to let go. The print will either
detach from the print bed, from the raft, or the print layers will
delaminate.

nordurljosahvida, 2020-01-02:
Absolutely, but in very many cases a raft with sufficiently small contact z distance has saved my prints, i too would love to know why, or how to identify that stress you're mentioning.

sungdzda, 2020-01-02:
Another scenario that ordains the use of rafts is when the model has a small contact area with the platform. Many new Prusa users would happily leave their printers alone and go to sleep only to find out 8 hours later a hotend/fan duct surrounded by plastic goo that is very frustrating and it is nearly impossible to undo the damage. Despite raftless printing (including any use of glue stick and surface treatment) having certain merits, in the generalized case a raft is the best option to ensure consistency, dimensional accuracy, and for preserving surface quality of the first layer.
[image]Image source reddit

bryanmtdt, 2020-04-28:
Please, this is an important setting, support and raft settings to be separated and first model layer height on other values rater than nozzle diameter, I've spend several hours trying to get a good balance but it seems to be pretty difficult right now.

alfredone78, 2020-08-31:
Please, fix this issue: I'd like to have a first layer (above the raft) with an height different from the nozzle size! For example, I want that the first layer is 0.2 mm, just like the other layers (default height).
Instead, now as things stand, I can't!!!

Zaniix, 2020-12-11:
I just ran into this issue as well. It does not seem to matter what settings I change, the first layer after the last raft layer is printed too high as if I was just printing in the air.  I wasted a lot of filament before I realized it was the slicers fault and not my printer.

kb-elmo, 2021-02-01:
same here.

layer height set to 0.2mm and contact-z distance set to 0.2mm as well I would expect the slicer to leave out exactly one layer between the last interface layer and the first object layer.

Instead I get three missing layers resulting in a gap of 0.6mm which results in mid-air printing and thus unusable print results.

Drachenkaetzchen, 2021-02-05:
I ran into the same issue. Please either fix this issue or at least make an in-slicer warning that this feature is currently broken. I spent hours debugging my hardware when in fact it was a slicer issue :(

nemart69, 2021-02-14:
Just FYI: I submitted PR #6003 which addresses most of the issues mentioned here a couple of days ago. This PR seems to be processed by PrusaSlicer development team.

Some examples here (white: Polymaker PC-Max: nozzle 0.4mm, 280/110 C in enclosure, silver: Prusa PLA):

[image]
[image]
[image]
[image]

bubnikv, 2021-02-24:
Implemented with #6003 and aee136c

Object over raft is no more printed with bridging flow, but with a normal flow.
Gap between the raft and 1st object layer is configurable independently from the support gap.

closed 2021-02-24 by bubnikv