prusa3d/PrusaSlicer #3427: čistící věž s jedním extruderem

joka-cz, 2019-12-30:
### Version
_2.2.0 Alpha 2 win64_

### Win 10 Pro 64bit CZ

### MK3S

### Behavior

Pokud ponechám zapnutou čistící věž při tisku s jedním extruderem, tiskárna se chová při manipulaci s filamentem jako kdyby byl přítomen MMU.

Bez čistící věže tiskárna "utrousí" kapku filamentu na místě, kde byla vyvolána výměna filamentu i když se jinou barvou začne tisknout úplně někde jinde.

Toto chování je už z dřívějších verzí 1.4.x

bubnikv, 2020-01-02:
Čeho chcete dosáhnout? Co je problém?

joka-cz, 2020-01-02:
dobrý den,

 

čistící věž při tisku MultiMateriál tisk s jedním extruderem se zřejmě chová jakoby byla přítomna jednotka MMU, ale tu já nemám. Pokud bych ponechal čistící věž, dochází k nárazu do nahromaděného kopce filamentu, nehledě na to, že extrudér v čistící věži přeskakuje, protože nemůže vytlačovat filament.

Při ruční výměně filamentu se bohužel tiskárna chová tak, že si pamatuje pozici, došlo k pokynu výměny filamentu, poté se tiskárna vrátí na pozici před výměnou a zde zanechá kapku druhé barvy a přesune se na pozici, od které se správně má tisknout druhou barvou.

 

Takže jsem předpokládal, že by se čistící věž použila k očištění trysky a poté by tiskárna najela na novou správnou pozici pro tisk druhé barvy. Že chování čistící věže přizpůsobíte volbě MultiMaterial tisk s jedním extruderem.

 

Bohužel můj předpoklad byl mylný, a tak musím při zapnutí více extrudérů vypínat čistící věž (automaticky ji zapínáte), a po výměně filamentu se pokusit odstranit nežádoucí artefakt tisku na pozici, kde byl vyvolán příkaz M600. A to nesmím dát do vlastního G-kódu v sekci pro výměnu materiálu dát příkaz M600.

 

Snad jsem to popsal trochu lépe.

 

Děkuji a zdravím

 

                Josef Karpíšek

rtyr, 2020-04-21:
Tohle souvisí s tím, co jsme probírali tady https://github.com/prusa3d/PrusaSlicer/issues/3428, parametr pro tisk jedním extruderem není určen k tomu, jak to používáte vy (M600 v toolchange skriptu na výměnu filamentu). Pro vaše použití nemá smysl mít čistící věž zapnutou. V našich profilech je zapnutá automaticky, protože se při zapnutém parametru pro tisk jedním extruderem počítá s tím, že používáte MMU. Tohle si myslím můžeme zavřít.

closed 2020-04-22 by rtyr