prusa3d/PrusaSlicer #3428: nadbytečný příkaz výměny nástroje

joka-cz, 2019-12-30:
### Version
_2.2.0 Aplha 2 win 64bit_

### Operating system type + version
_Win10 Pro 64 bit CZ_

### MK3S

### Behavior
při nastavení tisku více barev jedním extruderem je do G-kódu vložen na začátek příkaz pro výměnu nástroje, tzn. že po kalibrační čáře se ihned vyžaduje výměna nástroje (v případě M600 vysunutí a nové zavedení filamentu).

Prosba: nešlo by automaticky při nastavení tisku více barev jedním extruderem ve vlastnostech tiskárny dosadit M600 pro výměnu nástroje?

bubnikv, 2020-01-02:
Bohužel nerozumím. Můžete to prosím rozvést, pokud možno v příkladech? Co myslíte "více barev jedním extruderem"? MMU2? A nebo manuální výměnu filamentu?

joka-cz, 2020-01-02:
Zdravím,

 

mám na mysli manuální výměnu filamentu – multimateriál tisk s jedním extruderem

a Vlastní kód à G-Code pro výměnu nástroje M600

po vygenerování je po vytištění čistící čáry (nebo jak ji říkáte, v levé spodní části tiskového plátu – INTRO LINE) je ihned výměna nástroje, což ve verzích 1.4.x nebývalo a musím ten kód M600 ručně v editoru smazat. V příloze je soubor 3mf a gcode v ZIPu.

 

G1 Y-3.0 F1000.0 ; go outside print area

G92 E0.0

G1 X60.0 E9.0 F1000.0 ; intro line

M73 Q0 S23

M73 P0 R23

G1 X100.0 E12.5 F1000.0 ; intro line

G92 E0.0

M221 S95

G21 ; set units to millimeters

G90 ; use absolute coordinates

M83 ; use relative distances for extrusion

M600

T0

M104 S215 ; set temperature

M900 K30 ; Filament gcode

;BEFORE_LAYER_CHANGE

G92 E0.0

;0.2

 

Děkuji a přeji hezký zbytek dne

 

                Josef Karpíšek

rtyr, 2020-04-18:
M600 se za intro line nevkládá a pokud vím, ani to tak nikdy nebylo. Spíš se asi stalo to, že jste omylem při manipulaci s posuvníkem v preview kliknul hned u první vrstvy na "+", címž se vložila první výměna hned na začátek tisku. Teď už je to ošetřeno tak, že u první vrstvy není ikona "+" zobrazená. Zavírám.

joka-cz, 2020-04-18:
Dobrý den,

 

škoda, že jste si nepřečetli popis chyby, chyba je stále i ve verzi 2.2.0.

 

Teď jsem zkusil slicovat a je na 56. řádku příkaz M600. Můžete si to zkusit, přikládám 3MF soubor.

 

Josef Karpíšek

rtyr, 2020-04-18:
Já jsem si to přečetl, ale chápal jsem to tak, že používáte výměnu po vrstvách (kam PrusaSlicer automaticky vkládá M600). Asi máte na mysli něco jiného, zřejmě M600 v toolchange skriptu. Bylo by fajn, kdybyste přiložil 3MF projekt, aby bylo hned jasné, o co se jedná a nemuseli jsme to takhle lovit. Ani teď tu ten soubor s projektem není...

joka-cz, 2020-04-18:
při odpovědi přes email se soubor nevložil, napravuji takto
[Kryt baterie foto v5.zip](https://github.com/prusa3d/PrusaSlicer/files/4497082/Kryt.baterie.foto.v5.zip)

lukasmatena, 2020-04-18:
@joka-cz Oceňuji Vaši kreativitu při nastavování custom gcodů, jen je třeba to dotáhnout do konce. PrusaSlicer si myslí, že máte multimateriál. Výchozí stav u multimateriálu je, že není zaveden žádný filament. Proto se tam před tiskem objeví ta výměna, která je z Vašeho pohledu navíc. A protože v toolchange gcodu máte M600, je tam tedy mimo normální příkaz Tn také M600.

Je pravda, že i u MM se zpravidla filament loaduje ve start gcodu a v takovém případě je tam ten příkaz také svým způsobem navíc - toto by asi stálo za to vyřešit. U Vás nicméně ve start gcodu nic takového stejně není.

Váš problém by mohl jít vyřešit změnou custom toolchange gcode na toto: `{if previous_extruder!=-1}M600{endif}`. (Ta -1 popisuje stav, kdy tam zatím nic není.) Jestli a jak to bude fungovat si nejsem jistý, nevím, jestli tuto situaci vůbec někdo testoval, i když uznávám, že dává smysl. @rtyr mě dost možná v něčem doplní nebo opraví.

rtyr, 2020-04-18:
Já jen doplním, že zřejmě nebyl úplně správně pochopen význam parametru "multimaterial tisk s jedním extruderem". Je pravda, že ten název možná není úplně vypovídající. Jsou tím myšleny tiskárny, které používají jednu trysku/hotend, ale tisk probíhá stále v běžném multimaterial režimu, používají se klasické gcode příkazy pro změnu na konkrétní nástroje. Typický příklad je MMU upgrade pro Prusa i3. Tady tedy nemá žádné opodstatnění automaticky vkládat M600 (viz prosba v popisu issue), naopak to tam pro opravdové multimaterial tiskárny být nesmí. M600 není gcode na změnu nástroje, je to gcode na výměnu filamentu.

joka-cz, 2020-04-18:
děkuji Vám pánové za odpovědi

toto není moje myšlenka, ale pro mne téměř 2 roky stará informace z fóra a byla tam popisována technika a tehdy to fungovalo dobře - kromě mé osobní zkušenosti z té doby i zušenosti z posledních dvou příspěvku vlákna: [https://forum.prusaprinters.org/forum/original-prusa-i3-mk3-czech-cesky-jak-to-vytisknu-pomoc-s-tiskem/vice-barev-v-jedne-vrstve-tisku/#post-99685](url)

Nepoužívám to denně, ale pak jsem to opět využil a najednou jsem koukal, že tam ta M600 je z mého pohledu navíc. Na základě zkušeností z minulosti a nynějšího stavu jsem to reportoval jako možnou chybu.

Možná jsme využili vlastnost programu, kterou jste zamýšleli jinak (jak je ve Vašich odpovědích).

lukasmatena, 2020-04-19:
Máte pravdu, ještě ve verzi 2.0.0 se skutečně při první výměně nástroje vloží jen Tn, ale ne custom toolchange gcode. U ostatních se pak už vkládá. Z mého pohledu bylo tedy chybné původní chování, kdy byl custom toolchange gcode ignorován, přestože PrusaSlicer dělal výměnu. Byla tu na to i nějaká issues, např. #1245, #1530 a možná i další.

joka-cz, 2020-04-19:
Z mého pohledu první nástroj netřeba měnit po introline - ale to je ten úhel pohledu a využití vlastností programu.
Není to kritické, ostatní chování mně vyhovuje, tak prostě odmáznu první M600 v gcode v textovém editoru, když potřebuji tisk více barev v jedné vrstvě.
Pokud to tedy neberete jako chybu, můžeme to uzavřít.
Co jsem se setkal před x lety se Slic3r, udělali jste na něm obrovský kus práce a chápu, že je potřeba držet nějaké mantinely chování programy a požadavky od nás uživatelů zapracovávat s rozvahou, aby nenastalo nechtěné chování programu.

closed 2026-09-29 by prusaslicer-bot[bot]