prusa3d/PrusaSlicer #3430: PrusaSlicer crash on right-click on wipe tower

Grismo, 2019-12-30:
### Version
2.2.0-alpha2 +Win64

### Operating system type + version
Win7 Professinal SP1 64Bit

### 3D printer brand / version + firmware version (if known)
MK2.5S MMU2S Stock model / firmware 3.8.0

### Behavior
Upon clicking right at the wipe tower with MMU2S printer profile loaded PrusaSlicer crashes immediately. That happens when a model is loaded or even without a loaded model. In PrusaSlicer 2.2.0 that does not happen.

I tried the right click because my wipe tower was loosing adhesion to the print bed and I thought that maybe I can add brim to the wipe tower (what at least in Slicer 2.2.0 does not work).

lukasmatena, 2019-12-30:
I was able to reproduce it on my Linux setup. The crash happens when the tower is selected and it occurs in `Plater::priv::can_reload_from_disk()`. The wipe tower is represented by GLVolume that has no ModelVolume counterpart and it probably needs special treatment there. Probably something for @enricoturri1966 .

Thanks for reporting this.

bubnikv, 2020-01-24:
It is fixed on master, it will be part of the next alpha.

closed 2020-01-24 by bubnikv