prusa3d/PrusaSlicer #3432: Inconsistent / Non-Concentric Brim Patterns on PS 2.2.0 Alpha

AndrewSink, 2019-12-31:
### Version
_PrusaSlicer 2.2.0 Alpha_

### Operating system type + version
_MacOS Catalina 10.15.2 _

### 3D printer brand / version + firmware version (if known)
_Stock MK3S_

### Behavior
When slicing a model with a brim, the brim will be created with non-concentric  / non-linear toolpaths. The same model loaded in 2.1.0 will behave as expected.

* _Expected Results_
[image]
*Model sliced in 2.1.0, showing continuous toolpaths

* _Actual Results_
[image]
*Model sliced in 2.2.0, showing inconsistent toolpaths

*This error occurs with most models that have a curved or round bottom layer, even the Benchy included for print diagnostics 

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[3DBenchy.zip.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/4012649/3DBenchy.zip.amf.zip)

bubnikv, 2020-01-02:
This is an excerpt of the 2.2.0-alpha1 change log:

> The skirt is newly allowed to overlap with the brim, so that the skirt used as a wind draft shield is stabilized by the brim. Brim lines are split at the intersections so there is no overextrusion #724.

Does it answer your issue?

lukasmatena, 2020-01-25:
No response and the behaviour seems correct. Closing.

closed 2020-01-25 by lukasmatena