prusa3d/PrusaSlicer #3435: Add number of build instances to 3mf Metadata

nick-parker, 2019-12-31:
### Version 2.0.0+linux64-201905201652

### Operating system type + version
Ubuntu 16.04

### 3D printer brand / version + firmware version (if known)
Original Prusa Mk3s

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
I'm building an internal print logging app for a few Prusa machines and parsing the 3mf files to find the model files and number of copies being printed. That data is almost there in `Slic3r_PE_model.config` but not quite, forcing me to parse the (sometimes very large) `3dmodel.model` to access it.

Adding a Metadata tag for copies to each object tag would neatly solve my use case and probably be useful for many others.

Right now the objectid's increment by the number of copies between each object, but that doesn't work for the last object in the metadata. The attached file demonstrates the issue.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
[example.zip](https://github.com/prusa3d/PrusaSlicer/files/4013583/example.zip)

enricoturri1966, 2020-01-02:
@nick-parker

It was actually impossible to add a new metadata item into Slic3r_PE_model.config because that would break backward compatibility with older versions of PrusaSlicer.
I added the required information as an extra attribute of the object definition.
This is how your Slic3r_PE_model.config is saved now:

<?xml version="1.0" encoding="UTF-8"?>
<config>
 <object id="1" instances_count="1">
  <metadata type="object" key="name" value="d6b746e1-e853-464e-9d0e-df859887eed6.stl"/>
  <volume firstid="0" lastid="463">
   <metadata type="volume" key="name" value="d6b746e1-e853-464e-9d0e-df859887eed6.stl"/>
   <metadata type="volume" key="volume_type" value="ModelPart"/>
   <metadata type="volume" key="matrix" value="1 0 0 0 0 1 0 0 0 0 1 0 0 0 0 1"/>
  </volume>
 </object>
 <object id="2" instances_count="4">
  <metadata type="object" key="name" value="6744a50f-483d-42d4-acd7-2dd440cb26dc.stl"/>
  <volume firstid="0" lastid="5785">
   <metadata type="volume" key="name" value="6744a50f-483d-42d4-acd7-2dd440cb26dc.stl"/>
   <metadata type="volume" key="volume_type" value="ModelPart"/>
   <metadata type="volume" key="matrix" value="1 0 0 0 0 1 0 0 0 0 1 0 0 0 0 1"/>
  </volume>
 </object>
</config>

nick-parker, 2020-01-02:
That's perfect, thank you!

lukasmatena, 2020-01-31:
Implemented by 3d5e32a, as mentioned above. The improvement will be in the next release (2.2.0-alpha3). Closing the issue.

closed 2020-01-31 by lukasmatena