prusa3d/PrusaSlicer #3439: [Feature request] Enchanced modifiers (cooling, tempeartures, UI, etc)

guffy1234, 2020-01-01:
### Version
Version 2.1.1 win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Epo3D+

### Behavior
1. It would be nice to add cooling and probably temperature settings to set of settings available to modifier bodies. And probably it would be nice to have ability to use any settings that have a sense to some local area.
 
2. UI expiriense  of bodies managment list should be improved. Pls add right click menus, ability to delete a body or some subset of settigs (like removing entire "infill" subset instead of removing each setting one by one)

bubnikv, 2020-01-02:
> It would be nice to add cooling and probably temperature settings to set of settings available to modifier bodies. And probably it would be nice to have ability to use any settings that have a sense to some local area.

I agree, but it will take some work to fit it into our current architecture.

> UI expiriense of bodies managment list should be improved. Pls add right click menus

menus for what and where?

> ability to delete a body

delete button

> or some subset of settigs (like removing entire "infill" subset instead of removing each setting one by one)

I agree with this one, we should add such an option.

guffy1234, 2020-01-02:
Ideally cooling control should be comparable by flexibility as in IceSL slicer from french guys

mgx0, 2020-12-19:
Bump

Having a modifier (box, sphere and so on) that allows me to override cooling ould be extremely useful. Consider image below as an example. Now I have to blow air on the part by external fan to make sure the layers don't curl

[image]

treyd3, 2021-02-10:
Adjustment of part cooling via modifier/per-part settings is one of the only features that still keeps me from permanently deleting S3D. I print some fairly complex parts and I would so much rather have more control over part cooling than yet another infill pattern that I will probably never use.

Another reason to have this is that the automatic settings run the fans at higher speeds only DURING the moves that meet the criteria. Because fans don't speed up instantaneously, things like short overhanging loops never get the intended airflow - the fan will never even spin up to 100% RPM before the move is over and PWM drops back to the default.

Please add this functionality to 2.3.1!
Thanks!

robustini, 2021-03-19:
This is a key feature, why hasn't it been implemented yet?

mgx0, 2021-03-19:
it is a key feature, but looks like not for Prusa Research.

we need this feature because:
- PrusaSlicer does not detect overhang angles like Simplify3D does
- it does not have enforcer for fan speed
- changing fan speed manually on printer does not help because in gcode there is a lot of "intelligent" fan speed changes that turns down the fan immediately

PrusaSlicer is a lot better than S3D in many ways (and not because it's free. I have a paid S3D license which I don't use anymore), but their customer attitude makes me crazy sometimes. Not sure which motto applies here. "We have your money, thanks" or "don't tell us how to improve our product that is free of charge".

I know there are lot of users using PrusaSlicer on non-Prusa printer (because non-Prusa printer is priced at 20% of ORIGINAL Prusa printer), but Prusa Research should be aware that Prusa owners deserve added value for that price. Either make it paid or whatever, just do something. Or close issues that you don't plan to fix, so people can move on. This issue has been opened for more than a year and I don't believe that coronavirus or Trump is the culprit.

robustini, 2021-03-20:
> it is a key feature, but looks like not for Prusa Research.
>
> we need this feature because:
>
>     * PrusaSlicer does not detect overhang angles like Simplify3D does
>
>     * it does not have enforcer [...]

Totally agree!

stalawina, 2021-06-03:
Bump I want this to! I go back to S3D now only because I can control the cooling. Supports are also better in S3D, but that's another matter.

With the modifier features, Prusa Slicer has so much more potential than S3D regarding cooling control. S3D can only control it as a per layer basis, while Prusa Slicer has the potential of pinpointing any geometry for increased cooling using the part modifiers.

Please implement it!

OliveTheFemboy, 2023-03-12:
This still desperately needs implementation.

Another place where this is useful is when perimeters become overhangs and need something to build on. By printing perimeters inside to out (add this too as a modifer would be ideal), the perimeter on the inside of this hole can be built by building off the internal perimeters, but it needs significantly more cooling (You cannot simply increase cooling for the entire part, because layer adhesion becomes a problem and makes the part desperately weak) and a very slow speed to do so sucessfully.

This is a similar overhang printing technique to arc overhangs at it's core. It's just internal arcs, rather than external.

[image]
[image]

Tupson444, 2026-09-18:
still relevant