prusa3d/PrusaSlicer #3440: [Request] [SL1] Add Supports in slice view

mvasilakis, 2020-01-01:
### Version
_2.2.0 alpha2_

### Operating system type + version
Mac OS 10.14.6
### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Request
It would be useful in PS to add or move support from the Preview (⌘6) screen. 

Sometimes support is not in the ideal position or I'd like to add another support in the same area but once we switch to the 3D editor view (⌘5) it's impossible to find that spot again. So adding a support point in the Preview (⌘6) screen would make properly slicing a complex object infinitely easier.  

_Is this a new feature request?_
yes

#### Photo
[Screenshot 2020-01-01 17.19.34.pdf](https://github.com/prusa3d/PrusaSlicer/files/4014606/Screenshot.2020-01-01.17.19.34.pdf)

bubnikv, 2020-01-02:
> So adding a support point in the Preview (⌘6) screen would make properly slicing a complex object infinitely easier.

You want to edit the support points in the preview? Or just show the points there?

mvasilakis, 2020-01-02:
> You want to edit the support points in the preview? Or just show the points there?

I'd like to Edit in the ⌘6 screen. 
When you switch back to 3D editor view (⌘5) it's near impossible to find the spot again on a complex model. Moreover when you switch from ⌘6 to ⌘5 there is a view shift as well further complicating the issue. I've run into quite a few instances where I can't identify the area in the ⌘5 screen. Then I have to guess and wait for the complex model to reslice, see where I placed the support, nudge it, reslice, wait, repeat.

closed 2026-09-29 by prusaslicer-bot[bot]