prusa3d/PrusaSlicer #3443: Marlin FW: acceleration confusion in printer settings

GarthSnyder, 2020-01-02:
### Version
PrusaSlicer 2.2.0-alpha2+

### Operating system type + version
macOS Catalina 10.15.2

### 3D printer brand / version + firmware version (if known)
Evnovo/Artillery Sidewinder X1 (Marlin 2.0 firmware)

### Behavior
This is a section of the Machine Limits panel within Printer settings, for a printer whose firmware type is set to Marlin:

[image]

The first four items are indeed maximum accelerations and become arguments to M201 in the generated G-code. However, the last two -- "Maximum acceleration when extruding" and "Maximum acceleration when retracting" become arguments to M204, which sets _default_ accelerations. These values have nothing to do with maximum acceleration.

There are really several related issues here, so at the risk of pedantry, let me break this down just to be sure I cover them all.

1 - Marlin has separate default accelerations for printing, traveling, and retraction ([ref](http://marlinfw.org/docs/gcode/M204.html)). Most firmwares lump printing and traveling together and allow only two values for M204. For simplicity, PrusaSlicer is presenting Marlin as if it followed the general G-code standard, which is fine; that's not what I'm griping about.

The issue is that when you have one acceleration for "retracting", the other acceleration isn't for "extruding": it's for everything else. The current wording is actively wrong in that the "extruding" acceleration will be used to set the default travel speed, i.e., _the speed when not extruding_.

**Fix:** It would be clearer and more unambiguous to label these options something like "Default acceleration for retraction" and "Default acceleration for printing and traveling". 

2 - As mentioned above, these are default accelerations, not maximum accelerations.

**Fix:** Change "Maximum acceleration..." to "Default acceleration...". 

3 - ...but even if the wording is corrected, the options are still part of a settings group titled "Maximum accelerations".

**Fix:** Change the group title to "Accelerations" or break these items out into a separate "Default accelerations" group.

4 - OK, now the printer-based default accelerations are properly labeled. But now they're in apparent conflict with Print Settings -> Speed, which _also_ has a "default acceleration" setting. (Well, they were always in implicit conflict; now it's just more readily apparent.)

It appears that PrusaSlicer uses the Print Settings specification, if it exists, in preference to the Printer Settings values.

**Fix:** It seem like the current behavior is what you'd expect and want. Perhaps clarify in the Print Settings -- e.g., "Default (overrides printer settings)" -- so that someone who notices that the acceleration is set in two places can determine which value is actually used?

_Is this a new feature request?_
No.

bubnikv, 2020-01-02:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/1089 and https://github.com/prusa3d/PrusaSlicer/issues/1234

eschulma, 2020-05-13:
I ran into this same issue very painfully, see #4237 .

bubnikv, 2021-07-29:
PrusaSlicer 2.4.0-alpha1 will implement M204 PRT the Marlin 2 way for the new Marlin 2 firmware flavor, thus we have split the acceleration while extruding from the acceleration while traveling. Please test with the upcoming PrusaSlicer 2.4.0-alpha1 and let us know whether you are fine with the solution.

closed 2026-09-29 by prusaslicer-bot[bot]