prusa3d/PrusaSlicer #3444: empty layers detected error only on 2.1.1 not on 2.0

lt72884, 2020-01-02:
### Version
PS2.1.1

### Operating system type + version
Windows 10 Version 10.0.17763 Build 17763

Tevo tornado

### Behavior
* _Describe the problem_
I have made sure to try and replicate the issue in Prusa Slic3r v2.0, cura 4.4, S3d 4.0 BUT it only happens in PS2.1.1.
Issue: When i load a professionally made STL i paid for into PS2.1.1, put the layer height to 0.08mm, it errors out stating there are empty layers. I then tried it in version 2.0 of prusa slicer and it worked just fine. I even tried in other slicers with no issues:) If i switch the layer height from 0.08 to 0.10, no error. I do think this is a bug since it does not happen in version 2.0 of PS nor does it happen in other slicers. I do not want to just switch it to another layer height, i need it to print in 0.08mm please.  I rely on the high resolution for my engineered parts for prototyping

thank you for your time and concern. It is much appreciated

bubnikv, 2020-01-02:
Please attach the files to reproduce.

oleost, 2020-01-02:
I also have an issue with PrusaSlicer 2.1.1 for windows generating "Empty layers"

If I take the stock profiles, slice it it works without an error.
If I enable "Supports" and take 3 Rafts its gives me this error:

[image]

[image]

[Apple_watch_dock--002.zip](https://github.com/prusa3d/PrusaSlicer/files/4016554/Apple_watch_dock--002.zip)

lt72884, 2020-01-02:
As per requested, here is the file i used. 

thanks
[FDG-Skeleton_07_Sword.zip](https://github.com/prusa3d/PrusaSlicer/files/4016654/FDG-Skeleton_07_Sword.zip)

bubnikv, 2020-01-03:
Please test with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha2
It seems to me that the issue was fixed already. If not, then please attach the 3mf, which contains all the settings.

lt72884, 2020-01-03:
Ok, i will test out that version and report back to you.

lt72884, 2020-01-03:
I have had to set my extrusion multiplier to 3, to even get filament to
flow... never had to do this before. 3 is 300% flow rate.. i blame it on
windows 10

lt72884, 2020-01-04:
i just tried the alpha version as per requested and same error. here is a
screen shot and also my config ini files

lt72884, 2020-01-04:
[prusa slic3r tevo tornado.zip](https://github.com/prusa3d/PrusaSlicer/files/4021435/prusa.slic3r.tevo.tornado.zip)
[image]

lt72884, 2020-01-04:
I just tested a basic cube from version 2.1 and 2.2. no extrusion unless i set extrusion multiplier to 3. Do i need to make another bug report for that issue?

MastaRob, 2020-01-21:
Any news on this one? I have the same issue. Actually I tried 2.2.0 alpha and when I sliced initially, no problem so I thought it was fixed - then I imported my config bundle and it is now giving me the same error. Can attach as requested.
Its a problem for me with models that have a sharp point on top, one model I made I have to cut a little bit off the top and then it sliced fine.

lt72884, 2020-01-21:
Nope, i have not heard hide nor hair from anyone regarding this issue or
another bug report i sent in about having to set the extrusion multiplier
to 300% to even get it to print.

I have now switched to ideamaker slicer until prusaslicer gets fixed.

rtyr, 2021-04-27:
It is sliced correctly with "detect thin walls" option disabled (PS2.3.1). Closing.

closed 2021-04-27 by rtyr

rtyr, 2021-04-27:
>bug report i sent in about having to set the extrusion multiplier to 300%

I printed g-code generated from your 3mf without issues.