prusa3d/PrusaSlicer #3446: Send to Printer should be the default

GarthSnyder, 2020-01-02:
### Version
PrusaSlicer 2.2.0-alpha2+

### Operating system type + version
macOS Catalina 10.15.2

### 3D printer brand / version + firmware version (if known)
Not really relevant, but Evnovo Sidewinder X1 connected through OctoPrint

### Behavior
When the model has been sliced, you're given a choice of Export G-code and Send to Printer (assuming that your printer is connected). It used to be that these were two equal-sized buttons, but now it's a giant Export G-code button and a tiny little Send to Printer button on the side that bears only a mysterious symbol:

[image]

My beef -- and this is purely subjective opinion, although I suspect that many others are in the same boat -- is that in my workflow, Send to Printer is by far the more common operation. I only need to export G-code when debugging G-code. Who saves G-code as a matter of course?

Generally speaking, the most common operation should be the largest and most prominently placed. In 2.2, it's hidden away in a button so small that Fitt's Law starts to come into play. Not only that, but the button is identified only with a generic "G-code going somewhere" icon that could describe Export G-code just as well as Send to Printer. (As a matter of fact, the same icon is used for both Export G-code and Send G-code in the File -> Export menu. And why is the same operation called Send to Printer in the Plater window and Send G-code in File -> Export?)

**Desired fix:** Return to equal-sized, text-only buttons or at least swap the positions so that Send to Printer is the large, explicitly named default and Export G-code is behind the mystery icon.

_Is this a new feature request?_
No

McPcholkin, 2020-02-20:
Confirm, i use octoprint in 99.9% and small "Send" button not comfortable for my setup.
As solution i can propose add to settings checkbox "Swap Send and Export buttons" (maybe in "Print host upload" section).

foreachthing, 2020-02-21:
> As solution i can propose add to settings checkbox ..

Optional is always better for both worlds.

Rob65, 2021-02-04:
This is still the case in 2.3.0
Next to the way too large export button, there are two small buttons to send to printer or send to SD card. It would be good if we had an option to swap these. So either have the Send to printer or export to SD card being the bigger button of the three.

closed 2026-09-29 by prusaslicer-bot[bot]