prusa3d/PrusaSlicer #3447: Pointer to developer documentation and SDK for 3DConnexion

GarthSnyder, 2020-01-03:
### Version
PrusaSlicer 2.2.0-alpha2

### Operating system type + version
macOS Catalina 10.15.2

### Issue

The release notes say

> "We were unable to get the drivers, API description or any other support from the 3Dconnection company. If you believe that we at Prusa Research could do better integrating the SpaceMouses devices, please bug the manufacturer to provide us with the necessary support. Otherwise, we are left with googling of what others managed to reverse engineer or do the reverse engineering ourselves, which is likely not the most efficient use of our time."

I haven't been down this path before myself, but there appears to be complete documentation and SDKs available for download [here](https://www.3dconnexion.com/service/software-developer.html). You will need to register, but it doesn't appear that there's any formal agreement needed. There's a developer forum and a developer support email address as well.

I'm guessing that you found all this but that there was some other issue that remains unresolved. What was the nature of the roadblock? Could you use another pair of eyes?

_Is this a new feature request?_
No

LaserBee, 2020-01-03:
I am thrilled that they are trying to include Spacemouse compatibility. I have noticed a couple issues:

1) It's really slow to move and rotate. It doesn't seem adjusting the speed in the Spacemouse settings makes any difference at all. 

2) Once the object is sliced, the navigation is broken for both Spacemouse *and* normal mouse. Oddly, as long as I *do not touch* the space mouse before slicing, navigation works fine for the mouse after slicing. 

Anyway, I look forward to it working! :)

bubnikv, 2020-02-04:
Thanks, the third member of our team won a lotter at 3DConnexion company and he received access to the drivers. Why does 3DConnexion play such a lottery is beyond me.

closed 2020-02-04 by bubnikv