prusa3d/PrusaSlicer #3448: MMU2S Single gcode attempts to load filament before hotend is heated

akf-was-here, 2020-01-03:
### Version
2.1.0+Win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3s + MMU2s / 3.8.1 (1.0.6)

### Behavior
When starting a print with the printer field set to "Original Prusa i3 MK3S MMU2S Single" the gcode generated attempts to load the filament to the hotend before it's heated causing the gears to grind on the filament in the MMU2s.  If I manually preheat the nozzle it works fine, if I edit the gcode and move "Tx" command after the M109/M190 commands it works fine.  

I believe this is related to the lockout preventing the extruder motor from sending filament down while the hotend is cold.  I see on a preheated nozzle that the extruder motor is tuning and grabs the filament as it's sent down by the MMU2s but in the cold scenario it's not turning.

#### Project File (.3MF) where problem occurs
Renamed 3mf:

[DHL_XMAS_Frog_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/4018514/DHL_XMAS_Frog_fixed.zip)

bubnikv, 2020-01-03:
@tryr Roman maintains our printer profiles.

rtyr, 2020-01-03:
I don't see any problem. Tx command only promts user to select filament and then prepare/load the filament above the bondtech gears. There is no need for preheat. Tc command will load filament to the preheated nozzle later. If you have issues with first "Tx" phase you are probably using incorrect firmware on the printer or MMU unit. I tried gcode generated from your 3MF and it worked OK as intended on MK3SMMU2S.

Make sure you are using proper firmware for the printer (MK3S) and MMU2 unit. You can also try to reset MMU2 unit to factory defaults. If problem persists, please contact our support on live-chat.

akf-was-here, 2020-01-03:
- I did a factory reset on the MMU2s and behavior did not change.  
- Here is the support menu printout of my FW versions:
[image]
[image]

- I issued a Tx by itself (from an Octoprint terminal) and see the same behavior:  1) With the Hotend Cold it jams up against the extruder gear without pushing it out enough to flag the IR sensor and grinds up at the MMU2 gear continually until it eventually gives up, retracts and tries again.  Often it then gets stuck on the ground out part of the filament and the MMU2 fails to push it beyond that.  2) With the Hotend Warm spins the extruder gear in concert with the MMU2 gear and grabs the filament as it comes down pulling it in until the IR sensor trips.  
- I tried loosening the tension screw on the extruder door as much as possible to make it easier to push out and trip the sensor but no change in behavior.
- I can push the sensor flag in manually and make the grinding stop, then with the Tc happens later it completes the load fine.
- Again loading from the FW menu works perfectly fine (but it requires a preheat as part of the process) and loads for an MMU2 (NOT Single) work fine, even the initial load starting from Hotend Cold works fine.
- Looking at the gcode from the MMU2 (NOT Single) export I see that there is no Tx prior to the wait for temp commands (M109/M190), only a T0 (or what ever filament  you select T1, T2 etc) and this is AFTER the wait for temp commands (M109/M190) so it works fine.

When your Single mode load happens at cold, does your extruder gear spin?  If not spinning, Is the MMU2 capable of pushing hard enough to jam the filament between the gears anyway and trip the IR sensor then I guess?

I will contact live chat as you recommend.

rtyr, 2020-01-03:
Are you using stock bowden tube between MMU and extruder? (length). I tried both cold and hot scenarios on stock MK3S and in both cases MMU unit stopped feeding the filament "in time" even with disconnected ptfe from the extruder (no IR flag). I guess you can calibrate bowden tube length in MMU unit (see handbook), but it shouldn't be needed with stock PTFE tube. Gears are spinning in "hot scenario", you are right about that. I will send link to our FW devs, maybe we can improve it somehow or make it more consistent.

akf-was-here, 2020-01-03:
- Everything is stock, the MMU2s is so prone to issues I'm afraid to have anything non-standard.
- When I unscrew the bottom of the bowden from the extruder and issue the Tx it sends filament down to ~50mm past the end of the PTFE coupling and then slowly advances it by ~40mm at a time until it gives up at ~800mm, retracts ~1 bowden length, then starts stepping backwards ~40mm at a time until the mmu2 sensor detects a clear path and starts the process again.  After the 2nd attempt it gives up and flags an "MMU Load Failed" on the LCD. I can halt the send process by manually actuating the extruder door to force an IR sensor trip.  Note: these lengths are approximate because it  moves too often for me to get a super accurate measurement and there is a lot to physically juggle but I think you can get the general picture.

rtyr, 2020-01-04:
I am unable to reproduce this behaviour with Tx command. What about printing directly from SD card (not using Octoprint)? Same result?

akf-was-here, 2020-01-04:
You are correct, it's only present in Octoprint and I've traced it further to an add-on I was running to select the filament from the octoprint console rather than having to walk to my printer and manually select it:  OctoPrint-Mmu2filamentselect (https://github.com/derpicknicker1/OctoPrint-Mmu2filamentselect).

Sorry to have consumed your cycles, this is obviously not your responsibility. Should I go ahead and close this out?

rtyr, 2020-01-04:
No problem :-). Yes, I believe we can close this.

closed 2020-01-05 by akf-was-here