prusa3d/PrusaSlicer #3450: 3d Connextion Space Mouse "pivot" point, and not working after slicing

LaserBee, 2020-01-03:
### Version
LAtest. 2.2.0?

### Operating system type + version
_Windows 10

### Behavior
Love the Spacemouse implementation, but two issues. 

Firstly, the "pivot" point is halfway along the top edge of the build plate. Is it possible to move that to the centre of the plate? I think it would move more naturally.

Secondly, and more importantly, it doesn't seem to work after slicing. I can move the model around perfectly before I slice (though it is still a bit slow, even after CTRL-M'ing it), but once I slice, the space-mouse doesn't work, and the *normal* mouse is also wonky. 

Oddly, if I do *not* touch the spacemouse before slicing, then after slicing the normal mouse still works fine on navigation. If I use the spacemouse before slicing, then afterwards the spacemouse doesn't work at all, and the normal mouse is all messed up, rotating around an axis way off the build plate.

cosmith71, 2020-01-18:
I have these same issues.  Specifically, the Spacemouse doesn't work correctly on the Sliced Preview. 

I can replicate this by pushing the Spacemouse in a direction and holding it for more than a second or so in the Sliced Preview.  The pivot point, which is normally at the center of the plate, changes to a spot off of the plate (dependent on the direction of the Spacemouse push) for both the Spacemouse and the regular mouse in both the 3D Editor view and the Sliced Preview.  Holding the Spacemouse on the Sliced Preview in a different direction changes the pivot point to a different spot.

Once the pivot point is changed, the only way to get it back to the center of the plate is to close and restart PrusaSlicer.

2.2.0 Alpha 2
Windows 10

jltx1, 2020-01-20:
Maybe related, but after slicing I find that the space mouse is still working, just not updating the view.  If I move the regular mouse which forces a view refresh then the view snaps to the new position.  If I slowly move the normal mouse so the view is refreshing, then the space mouse mostly works, though it is fighting the normal mouse updates.

BTW, the object view works fine but I only have 5 axes.  The left/right roll axis does not function.

2.2.0 Alpha 2
Mac OS 10.13

bubnikv, 2020-03-15:
is it working for you now in PrusaSlicer 2.2.0-rc3?

cosmith71, 2020-03-15:
Seems to work very well for me now.

Thanks!

bubnikv, 2020-10-17:
thanks, closing.

closed 2020-10-17 by bubnikv