prusa3d/PrusaSlicer #3451: [request] Option to enable seperate wipe/prime tower for each extruder

mkudzia84, 2020-01-03:
Option to enable construction of seperate wipe/prime tower for each extruder.

This would be helpful for:
- true multi extruder setups
- MMU2s where materials don't adhere well to each other. (purging inside and then having more solid/layer adheering exterior to help the integrity of the tower)

Lukrative525, 2020-05-07:
This seems like a really good idea to me.

Quas7, 2020-10-07:
the user would just have to avoid collisions when approaching the two different heights of wipe towers after printing many layers with one material but not the other.

acekrystal, 2023-12-15:
I was hoping to use the Prusa XL for multi material support with incompatible materials.. It works perfect... was it not that the wipe tower almost alway's fails,

The wipe tower is now build as a sandwitch of incompatible materials. So an option to section incompatible materials, or have a dedicated wipe tower per material type seems a must.

masto, 2023-12-23:
Indeed, the situation is pretty gross when trying to mix PLA and PETG (per https://help.prusa3d.com/article/combining-materials-xl_498103#pla-petg). This (or #3070, I'm not sure of the difference) would be very useful to XL owners.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]