prusa3d/PrusaSlicer #3455: [request] Printing order for Sequential Printing

CNCKitchen, 2020-01-03:
### Version 2.2.0 alpha 2

I'd like to use PrusaSlicer more for my DOE prints for which I often use sequential printing. Unfortunately the printing order of my individual part seems kind of random which can cause print head interference. It would be great if the printing order was similar to the order of the parts in the part-list.

[image]

stuartpb, 2020-01-22:
It'd also be nice if there were some way to intelligently plan/sort a print order, like front to back (so the cable bundle doesn't start knocking parts off the plate) or shortest to tallest.

bubnikv, 2020-01-23:
We are working on it. The current master already prints in the order the objects and instances are shown  at the object list. We will also

1) Show tags over the objects in the 3D scene including object name and print order.
2) Allow one to reshuffle the objects in the object list by a drag & drop.

lukasmatena, 2020-02-06:
Both improvements mentioned above are implemented and will be in PrusaSlicer 2.2.0-alpha4 release.
Closing.

closed 2020-02-06 by lukasmatena