prusa3d/PrusaSlicer #3456: Try to make top/bottom layer lines continuous and straight wherever possible

yoyo42, 2020-01-03:
### Version
PrusaSlicer 2.1.1+win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Stock i3 mk3, 

### Behavior
Feature request: change to slicing algorithm for infill/solid areas:

Imagine a square with a hole in the centre. When printing this the perimeters of the square and the hole are done first, then rectilinear fill is flood filled into it. This leads to areas being done in long or short lengths ending at the hole, then eventually picking up again to fill the other side of the hole. I see visible artifacts due to this. If the infill could be treated as if the hole wasn't there, just retracting to skip it, then the overall effect would be much smoother.

This would also help speed up and smooth out things like lithophanes that are done with many perimeters to make them solid. The entire outer perimeters could be printed first then only a few linear lines with frequent retractions would be used to fill the centre (fast) instead of it printing dozens of tiny random shaped perimeters and having to fill them all individually (very slow).

I can generate a couple of test cases and sketches of the path I would like it to take if that helps.

espr14, 2020-01-08:
Nice idea. I think somebody already wrote this some time ago. In fact, this is even easier to code than splitting rectilinear pattern into pieces.

rtyr, 2021-04-27:
I believe this is solved by monotonic infill, even though it is not exactly the algorithm described here.

Closing.

closed 2021-04-27 by rtyr