prusa3d/PrusaSlicer #3459: alpha2 builds successfully, but links against liblibnest2d.so

jasontibbitts, 2020-01-03:
### Version
2.2.0-alpha2

### Operating system type + version
Linux - Fedora 31

### Behavior
The build succeeds and all the tests pass, but the resulting binary has an unsatisfied reference to a library "liblibnest2d.so".  Looking at the build lines, it appears that it tries to link that dynamically and build an rpath into the binary:

```
/usr/bin/c++  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fext-numeric-literals -Wall -Wno-reorder -pthread -DNDEBUG  -Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld CMakeFiles/PrusaSlicer.dir/PrusaSlicer.cpp.o  -o prusa-slicer  -Wl,-rpath,/builddir/build/BUILD/PrusaSlicer-version_2.2.0-alpha2/Build/src/libnest2d: libslic3r/liblibslic3r.a -ldl -lstdc++ slic3r/liblibslic3r_gui.a -pthread -lwx_baseu-3.0 -lwx_gtk3u_core-3.0 -lwx_gtk3u_adv-3.0 -lwx_gtk3u_html-3.0 -lwx_gtk3u_gl-3.0 /usr/lib64/libcurl.so /usr/lib64/libz.so -ldl libslic3r/liblibslic3r.a libnest2d/liblibnest2d.so /usr/lib64/libnlopt.so admesh/libadmesh.a -lminiz /usr/lib64/libboost_system.so /usr/lib64/libboost_log.so /usr/lib64/libboost_filesystem.so /usr/lib64/libboost_thread.so /usr/lib64/libboost_log_setup.so /usr/lib64/libboost_locale.so /usr/lib64/libboost_regex.so /usr/lib64
/libboost_chrono.so /usr/lib64/libboost_atomic.so /usr/lib64/libboost_date_time.so clipper/libclipper.a boost/libnowide.a /usr/lib6/libexpat.so glu-libtess/libglu-libtess.a polypartition/libpolypartition.a -lpoly2tri qhull/libqhullstatic.a -lm semver/libsemver.a /usr/lib64/libtbb.so -pthread avrdude/libavrdude.a imgui/libimgui.a /usr/lib64/libGLEW.so /usr/lib64/libGLU.so /usr/lib64/libGL.so hidapi/libhidapi.a -ldl
```

But it doesn't appear that the shared library is being installed, and using rpath like that doesn't work well if the binary is going to live in a location different than where it was built (though I guess it would be possible to modify the rpath after the fact).

I looked at the 2.1.1 build logs and the rpath manipulation just doesn't appear there at all and in fact I don't even see that a libnest2d shared library was built at all there..  So I'm guessing this has something to do with commit fbc174ad061a0c6dbbd2cc66455e4bd949517af3 which reworked how the libnest2d code is built, but I'm afraid I don't understand enough about how that changed things.

Any hints on how I might be able to tweak things to link in libnest2d more like it was in 2.1.1?

bnavigator, 2020-01-06:
Maybe related, maybe not:

On the OpenSUSE build there is no rpath but the test for libnest2d fails:

```
[ 2580s] [ 86%] Linking CXX executable prusa-slicer
[ 2580s] cd /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/PrusaSlicer.dir/link.txt --verbose=1
[ 2580s] /usr/bin/c++  -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -pthread -O2 -g -DNDEBUG  -flto=auto -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now CMakeFiles/PrusaSlicer.dir/PrusaSlicer.cpp.o  -o prusa-slicer  libslic3r/liblibslic3r.a -ldl -lstdc++ slic3r/liblibslic3r_gui.a -L/usr/lib64 -pthread -lwx_baseu-3.1 -lwx_gtk2u_core-3.1 -lwx_gtk2u_html-3.1 -lwx_gtk2u_gl-3.1 /usr/lib64/libcurl.so /usr/lib64/libz.so -ldl libslic3r/liblibslic3r.a libnest2d/liblibnest2d.so /usr/lib64/libnlopt.so admesh/libadmesh.a miniz/libminiz_static.a /usr/lib64/libboost_log.so.1.71.0 /usr/lib64/libboost_filesystem.so.1.71.0 /usr/lib64/libboost_locale.so.1.71.0 /usr/lib64/libboost_system.so.1.71.0 /usr/lib64/libboost_thread.so.1.71.0 /usr/lib64/libboost_regex.so.1.71.0 /usr/lib64/libboost_chrono.so.1.71.0 /usr/lib64/libboost_atomic.so.1.71.0 /usr/lib64/libboost_date_time.so.1.71.0 clipper/libclipper.a boost/libnowide.a /usr/lib64/libexpat.so glu-libtess/libglu-libtess.a polypartition/libpolypartition.a poly2tri/libpoly2tri.a qhull/libqhullstatic.a -lm semver/libsemver.a /usr/lib64/libtbb.so -pthread avrdude/libavrdude.a imgui/libimgui.a /usr/lib64/libGLEW.so /usr/lib64/libGLU.so /usr/lib64/libGL.so hidapi/libhidapi.a -ldl 
```

```
[ 2803s] [ 89%] Linking CXX executable libnest2d_tests
[ 2803s] cd /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build/tests/libnest2d && /usr/bin/cmake -E cmake_link_script CMakeFiles/libnest2d_tests.dir/link.txt --verbose=1
[ 2803s] /usr/bin/c++  -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -O2 -g -DNDEBUG  -flto=auto -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now CMakeFiles/libnest2d_tests.dir/libnest2d_tests_main.cpp.o CMakeFiles/libnest2d_tests.dir/printer_parts.cpp.o  -o libnest2d_tests  ../../src/libnest2d/liblibnest2d.so ../../src/clipper/libclipper.a /usr/lib64/libnlopt.so /usr/lib64/libtbb.so -pthread -ldl 
[ 2813s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build'
[ 2813s] [ 89%] Built target libnest2d_tests
```

```
[ 3297s] 1/5 Test #1: libnest2d_tests ..................***Failed    0.03 sec
[ 3297s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build/tests/libnest2d/libnest2d_tests: error while loading shared libraries: liblibnest2d.so: cannot open shared object file: No such file or director
```

For now i just skip that test in the %check section.

And there is the `-DLIBNEST2D_STATIC` parameter, but I am not sure if it does anything

jasontibbitts, 2020-01-15:
I just noticed c0b90f68ab54bb327227ad0e0aed947ca46c8837, which fixes this issue for me.

closed 2020-01-16 by jasontibbitts