prusa3d/PrusaSlicer #3461: Git Linux build failing RemovableDriveManager.cpp

clytle374, 2020-01-04:
On Gentoo from source, built fine last week.  Latest check is:

```
git rev-parse HEAD
9406b5044753cff3ed20ec28f6c5b6067e6fd5a3
```

Now failing like this. 
```

/home/cory/sources/prusa/src/slic3r/GUI/RemovableDriveManager.cpp: In member function ‘void Slic3r::GUI::RemovableDriveManager::inspect_file(const string&, const string&)’:
/home/cory/sources/prusa/src/slic3r/GUI/RemovableDriveManager.cpp:328:43: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
  328 |      std::string name = basename(path.data());
      |                                  ~~~~~~~~~^~
      |                                           |
      |                                           const char*
In file included from /home/cory/sources/prusa/src/slic3r/GUI/RemovableDriveManager.cpp:21:
/usr/include/libgen.h:34:36: note:   initializing argument 1 of ‘char* __xpg_basename(char*)’
   34 | extern char *__xpg_basename (char *__path) __THROW;
      |                              ~~~~~~^~~~~~
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:869: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/RemovableDriveManager.cpp.o] Error 1

```
Let me know if I can provide anymore details.
Thanks
Cory

clytle374, 2020-01-04:
commit 80ef0e033498307fdad73de50fd87607a10a9848  builds and runs

G4DCP, 2020-01-04:
Can confirm same problem trying to build on Rasp Pi 4.
-- 
Peter G4DCP

G4DCP, 2020-01-05:
commiit 30f7a2b8e also appears to be OK so the bad commit is -

9406b5044753cff3ed20ec28f6c5b6067e6fd5a3 is the first bad commit

Peter G4DCP

lukasmatena, 2020-01-05:
I can confirm this. On my system, `basename` is #defined in libgen.h and translates to `__xpg_basename`, which takes non-const `char*` as argument. That's why the build fails.

However, I doubt that casting-away `const` as it was is a solution - according to comments in libgen.h, the function "sometimes will modify its argument", and that's something one shouldn't do to std::string (although it probably depends on what exactly that function modifies). I bet that the safe solution is to pass a copy instead? @kocikdav @bubnikv

lukasmatena, 2020-01-05:
Actually, there appear to be two basename functions out there: see NOTES section in http://man7.org/linux/man-pages/man3/basename.3.html. The one in string.h takes const argument.

If I #include string.h instead of libgen.h, the code compiles. But I have no clue if it is safe to do so.

bubnikv, 2020-01-06:
Maybe we could use boost::filesystem for that instead.

clytle374, 2020-01-08:
Was trying to test the changes and now I'm getting the following. Not sure if I should file a separate bug report as this seems to be from the same batch of changes. 

`/home/cory/sources/prusa/src/libslic3r/EdgeGrid.cpp: In member function ‘Slic3r::EdgeGrid::Grid::ClosestPointResult Slic3r::EdgeGrid::Grid::closest_point(const Slic3r::Point&, coord_t) const’:
/home/cory/sources/prusa/src/libslic3r/EdgeGrid.cpp:1150:25: warning: comparison of integer expressions of different signedness: ‘size_t’ {aka ‘long unsigned int’} and ‘int’ [-Wsign-compare]
 1150 |  if (result.contour_idx != -1 && d_min <= double(search_radius)) {
      |      ~~~~~~~~~~~~~~~~~~~^~~~~
/home/cory/sources/prusa/src/libslic3r/Config.cpp: In member function ‘void Slic3r::DynamicConfig::read_cli(const std::vector<std::__cxx11::basic_string<char> >&, Slic3r::t_config_option_keys*, Slic3r::t_config_option_keys*)’:
/home/cory/sources/prusa/src/libslic3r/Config.cpp:767:47: error: invalid conversion from ‘const char**’ to ‘char**’ [-fpermissive]
  767 |     this->read_cli(int(args.size()), args.data(), extra, keys);
      |                                      ~~~~~~~~~^~
      |                                               |
      |                                               const char**
In file included from /home/cory/prusaslic3r/src/libslic3r/libslic3r_pch/pchheader.hpp:112:
/home/cory/sources/prusa/src/libslic3r/Config.hpp:1782:51: note:   initializing argument 2 of ‘bool Slic3r::DynamicConfig::read_cli(int, char**, Slic3r::t_config_option_keys*, Slic3r::t_config_option_keys*)’
 1782 |     bool                read_cli(int argc, const char* const argv[], t_config_option_keys* extra, t_config_option_keys* keys = nullptr);
      |                                            ~~~~~~~^~~~
/home/cory/sources/prusa/src/libslic3r/Config.cpp: At global scope:
/home/cory/sources/prusa/src/libslic3r/Config.cpp:770:6: error: no declaration matches ‘bool Slic3r::DynamicConfig::read_cli(int, const char* const*, Slic3r::t_config_option_keys*, Slic3r::t_config_option_keys*)’
  770 | bool DynamicConfig::read_cli(int argc, const char* const argv[], t_config_option_keys* extra, t_config_option_keys* keys)
      |      ^~~~~~~~~~~~~
`

lukasmatena, 2020-01-08:
I believe none of us had problems with this bit, including our build servers. What compiler and what version are you using? Honestly, I'm unable to tell if that code is somehow ambiguous or not. You may try changing the second parameter of `read_cli` to `const char* const* argv`, which I think is the intended meaning.

clytle374, 2020-01-08:
Current git builds and runs as it should.  

Sorry I screwed up there, ran make clean and problem went away. I thought I did that earlier

This issue is resolved.

lukasmatena, 2020-01-08:
Ok, I'm closing the issue.

closed 2020-01-08 by lukasmatena