prusa3d/PrusaSlicer #3462: I set up support enforcers and I get nothing. I am using the box stretched out to go from build plate to the bottom on the overhang.  I have "generate support material" checked.  "Auto generate" unchecked.  Version 2.1.0 Win 64.

OldSalt1945, 2020-01-04:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

audiorazor, 2020-01-05:
Did you put the drop down box for enforcers and blockers only on the plate tab?

OldSalt1945, 2020-01-23:
Sorry for the late response. I (insert lame excuse here). 
I chose (I'm fairly certain) "For support enforcers only".
The support encompassed build plate and structure.  Part of it should have been on the build plate and a portion of it would have rested on printed structure.
I am pretty new to 3-D, so I may have fouled something up.  As I recall, after banging my head on the problem, I just lived with the sagging threads and moved on to other parts I was printing.
 John

bubnikv, 2020-03-15:
please provide 3MF and screenshots.

rtyr, 2020-04-22:
@OldSalt1945 Trying again before closing. Can you provide 3MF? My guess is that you probably inserted support enforcer under the bridge. In that case you have to disable "Don't support bridges" parameter for supports to be generated.

rtyr, 2020-05-18:
No response. Closing.

closed 2020-05-18 by rtyr