prusa3d/PrusaSlicer #3465: version 2.0,2.1,2.2 alpha will not extrude unless multiplier is at 3

lt72884, 2020-01-04:
### Version
2.0 ,2.1, and 2.2 alpha

### Operating system type + version
Windows 10    Version	10.0.17763 Build 17763

### 3D printer brand / version + firmware version (if known)
Tevo Tornado with marlin
### Behavior
I have tried all versions of Prusa slicer extensively the last 4 days. I have triple checked all settings, but it will not extrude unless the extrusion multiplier is at 3. I have tried it in all versions. It also gives me empty layer error ALOT in version 2.1 and 2.2 alpha, but when it doesn't give empty layer error, it wont extrude(in any version). attached is a zip which contains my settings and a stl that sliced properly.

Version 2.0 USED to work.

I also tried this on multiple computers with exact same result. The other PC had exact same build version of windows 10..

[prusa settings.zip](https://github.com/prusa3d/PrusaSlicer/files/4022339/prusa.settings.zip)

bubnikv, 2020-03-15:
I suppose you have a 3mm filament dmr system, but you set the filament dmr in prusaslicer to 1.75mm?

lt72884, 2020-03-15:
Nope, i do not. Just a stock cr10 running 1.75mm pla.

rtyr, 2020-04-06:
Is this issue still valid? I don't think that this is slicer related. Also, you are mentioning Tevo Tornado in the inicial post, but CR10 in the later response.

rtyr, 2020-04-23:
No response, closing.

closed 2020-04-23 by rtyr