prusa3d/PrusaSlicer #3466: Avoid Crossing Perimeters leads to crash

mwjones1971, 2020-01-05:
### Version
2.1.0 + win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 64-bit Build 1909
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 68,632MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.41
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2070 SUPER/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
[... 342 lines cut ...]
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3S+MMU2S 3.8.1/1.0.6

### Behavior
Slicing an object that is near the edges of the build platform, if "Avoid Crossing Perimeters" is enabled, upon slicing the object, the resultant GCODE will move the X-axis beyond its limits, resulting in the extruder crashing, and the printer erroring.

Problem does not occur if "Avoid Crossing Perimeters" is disabled.

To reproduce, open the .3MF in the attached ZIP file and attempt to slice with "Avoid Crossing Perimeters" enabled. Preview all travel views, particularly at left front and right rear corners of the build plate, travel movements will exceed the build plate.

#### Project File (.3MF) where problem occurs

[flexicat_flat_ear.zip](https://github.com/prusa3d/PrusaSlicer/files/4022929/flexicat_flat_ear.zip)

nordurljosahvida, 2020-03-17:
same here with anycubic predator [delta printer], see attached 3mf.

[prusaslicer-avoid_crossing_perimeters-issue.zip](https://github.com/prusa3d/PrusaSlicer/files/4342947/prusaslicer-avoid_crossing_perimeters-issue.zip)

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

Tested with the models provided, it crashes no more.

closed 2020-12-11 by bubnikv