prusa3d/PrusaSlicer #3468: Select default extruder

sticilface, 2020-01-05:
### Version
2.1.0+

### Operating system type + version
10.14.6 (18G103) 

### 3D printer brand / version + firmware version (if known)
MK2.5S MMU2S

_Is this a new feature request?_

YES.  

currently extruder 1 is automatically the default.  I'd like to use extruder 5.  
why?  - my default colour is black.  I quite often like to print my more sensitive colours yellow, white etc, before the black so that these features do not have any 'black' appear in them. 
currently the default extruder does not appear to be configurable - in which case it should just say - extruder 1.  
[image]

I would like to propose that the 'default extruder' is selectable and should appear in the multiple extruder settings window. 
[image]

Thanks for all the hard work. 2.2.0 looks very promising!

ruedli, 2020-01-08:
I wonder if this is not already achievable, when I select for the "stl" (where is says default in the  extruder column) an extruder, in your case "5", them every object where you subsequently choose "default" will use and show extruder 5.

So... you have the option already: just set default at the .stl level to extruder 5 (so  this is is where you specify your desired default), and this specified extruder will be your default extruder for every object that subsequently chooses "default". Or is there more to consider?

koenkooi, 2020-01-09:
Sorf of related to this: there's no setting for selecting the 'brim' extruder, which means that for https://www.prusaprinters.org/prints/15346-prusa-mini-multicolor-front-y-plate it will select the white extruder for brim, not the black extruder :/

GarthSnyder, 2022-07-02:
Another reason to want this feature: if you have saved plates of items and need to print the batch regularly but with different extruders/colors, it's quite tedious to go through and set all the extruders by hand. I'd much rather just set one default extruder for the session and leave the saved configuration with everything set to "default".

jmd73, 2022-07-03:
@GarthSnyder In the meantime as a workaround, you can select all the objects on the bed and right click and select extruder for all at once.

Image just as an example:
[image]

GarthSnyder, 2022-07-03:
Thanks for mentioning that, @jmd73 - I had completely missed it! Yes, that's nearly as good. 

Given that, it's odd that this doesn't work in the object list pane. If you right click and pick an extruder with multiple objects selected, it only sets the extruder for the object that was directly under the right-click.

stepikovo, 2022-07-06:
@GarthSnyder which version are you using? On Win, with 2.4.2, it works just fine in the object list

GarthSnyder, 2022-08-18:
@stepikovo: yes, I was out of date. Works fine for me in current builds.

Back to the original issue... My primary application for multicolor printing is filament management. That is, I rarely combine colors in a single object. I just like to have multiple filaments on tap and be able to select the one I want to use without having to mess with reloading filament.

As a consequence, I tend to print extended runs of objects with the same filament as I'm creating, testing, and tuning new designs. If an object's going to be printed in, e.g., red PETG, I just want to be in the "red PETG" world where everything I print uses that filament by default.

It's not a big deal to select a body and hit a number key to select red PETG. What IS a big deal is _remembering_ to do that each and every time I run a print. I constantly (maybe 30% of the time) repeat the loop of forgetting this step, starting the print, realizing the printer is loading the wrong filament, aborting the print, resetting the filament choice, cleaning up the printer state, and restarting. _That_ is a real pain.

A good solution would be "not being such an idiot." But sadly, I'm not having much luck implementing this, so I'd really appreciate some help from PrusaSlicer.

The OP suggested a profile parameter that sets the default filament. That would be helpful and would solve my problem. But I think my use case could be covered by something even simpler, namely, just defaulting the filament for newly-added objects to be whatever filament was most recently assigned (to any object). It wouldn't have to persist across app executions.

I guess I also question whether the concept of a default filament makes sense for anyone. People with multi-filament systems are particular about their filament choices. Does anyone really print with an "I don't even want to think about what filament, just use the default" approach? More likely, they just make the default be whatever filament they use most frequently, to save keystrokes. But they still want to make an explicit choice of filament.

Maybe the default in multifilament mode should just be "no filament assigned," leading to an error if you try to slice or print.

scottnotrobot, 2022-11-27:
i'd imagine there are bigger fish to fry but fwiw i'd like to have this also so i'll throw out a me2pls

Razor412, 2024-12-10:
Just to be clear, this issue was never acted on correct? I found this request while searching for a way to set a different default extruder on my XL. @Ro3Deee is right on the money! ;-)

Razor412, 2024-12-11:
Can confirm this feature was never added to PrusaSlicer. Spoke to Prusa support about it today and their messaging was that it was the way that it was because of the single tool XL's. The rep agreed about the need for it though.

Chase-Money, 2025-03-18:
I would like to second the desire to set your default XL extruder to something other than being extruder 1 automatically. Not that its a big deal, I guess I'll just always load plain old black into Extruder 1 in the future. lol. So yes, there are bigger fish to fry. But it would improve Quality of Life for XL users in the same way being able to select certain extruders for infill does for instance. Not as big of a deal as being able to print infill with a larger nozzle obviously, but still. Love my XL btw!

closed 2026-09-29 by prusaslicer-bot[bot]