prusa3d/PrusaSlicer #3469: PrusaSlicer 2.2.0 alpha 2 change filmament at height command does not work with ender 3

pierremilcent, 2020-01-05:
### Version
PrusaSlicer 2.2.0 alpha 2

### Operating system type + version
Windows 7, win64 version

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro, marlin 1.1.9 with BLTouch

## Behavior
When addding a change filament command at slicing step, then the printer does not stop or anything to change the filament.

I expected this to work as it does with Cura 4.4.1 and "Pause at height" script.

bubnikv, 2020-01-06:
Could you compare the G-code generated with PrusaSlicer to G-code generated by Cura? What are the differences in the "pause at height" sequence?

tutulino, 2020-01-09:
PrusaSlicer change filament command uses M900 gcode, it only works if your printer has activated advanced pause feature in Marlin, Ender (and most printers) has disabled this feature by default.

You must use pause at height or change your firmware to Marlin and activate this feature

tigert, 2020-02-08:
Just for the record, filament change works with my stock Ender 3 Pro, software version 1.1.6.2.

It is likely your bltouch-enabled firmware has this disabled.

rtyr, 2020-04-21:
PrusaSlicer is using standard M600 command for filament change. Filament change support must be enabled in the firmware. Closing.

closed 2020-04-21 by rtyr