prusa3d/PrusaSlicer #3470: [2.2.0-alpha2] Wrong settings for Generic PLA on Ender-3

probonopd, 2020-01-05:
### Version

PrusaSlicer-2.2.0-alpha2+linux-x64-201912221342.AppImage

### Operating system type + version

Xubuntu 18.04

### 3D printer brand / version + firmware version (if known)

Creality Ender-3

### Behavior
* In the Wizard, select Creality
* Select Ender-3
* Select "Generic PLA"

The settings are wrong. These are not suitable for PLA.

[image]

This bug was originally discovered by Chuck Hellebuyck from the CHEP YouTube channel:

https://youtu.be/G_4H1OTIAK0?t=157

bubnikv, 2020-01-06:
@rtyr ?

rtyr, 2020-01-06:
It is OK in the configuration bundle. Probably some issue with symbolic names.

[image]

bubnikv, 2020-01-06:
Interesting, I don't see this problem.

probonopd, 2020-01-06:
Select "Generic PLA", __not__ "Creality PLA"

rtyr, 2020-01-06:
[image]

rtyr, 2020-01-06:
I managed to reproduce that issue. It seems like a bug in the filament wizard.

YuSanka, 2020-01-08:
Thanks for your report. The issue was caused by a missing visibility control for compatible presets. Thus in some cases you can see settings for the first compatible preset instead of the first compatible and visible preset. This bug is fixed and will be a part of the next release.

closed 2020-01-08 by YuSanka

probonopd, 2020-01-08:
Thank you very much.