prusa3d/PrusaSlicer #3471: Feature Request: Slope display

WarrenSchultz, 2020-01-05:
### Version
2.1.10

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Mk3S/MMU
Mini

### Behavior
Feature request:

When doing final visual checks before printing, it would really simplify the process if the "View" dropdown could have a color-coded view for showing the slope.
Ideally (at least to me), yellow would be the just beyond the support threshold, with gradations to red for most support required. (A slider in advanced mode to tweak that threshold in realtime would be perfect, but not required for an initial iteration.) Thanks!

blibenson, 2020-02-09:
This is a great suggestion, unfortunately, the support threshold support settings do not work correctly. The bug was posted last year in October and remains unfixed. The slicer pretty much ignores the actual value that the threshold is set at. For example you need to set the threshold to 57 degrees to get 45 degree overhangs supported.

bubnikv, 2021-10-07:
Implemented already at the paint-on supports gizmo.

closed 2021-10-07 by bubnikv