prusa3d/PrusaSlicer #3472: Round up time and weight for gcode file name values

neophyl, 2020-01-06:
### Version
All

### Operating system type + version
Not applicable

### 3D printer brand / version + firmware version (if known)
Not applicable

### Behavior
With the placeholders for generating gcode names I usually use {print_time}_{total_weight}.  With the limited functionality on scrolling names (either not scrolling on some printers or taking a long time to show all the characters on the Mk3) would it be possible to round up these values to the nearest gram and nearest minute please.  
It doesn't really make sense to have filament grams listed to 3 decimal places or to have the seconds in the time. 

Rather minor I know given the list of stuff already but hopefully an easy one :)

_Is this a new feature request?_
Yes

Carib3D, 2020-07-01:
I have the following output format:
   {input_filename_base}_{nozzle_diameter[0]}n_{layer_height}L_{filament_type[0]}_{print_time}_{total_weight}g.gcode
which comes out as, for example, 
   Spacer_0.6n_0.25L_PLA_1h16m_34.196g.gcode
So I don't get the seconds in time but I do get the weight to 3 decimals which I agree is a nuisance and would hope there is a way to round up.

neophyl, 2020-07-01:
I've been using the following
{input_filename_base}_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}_{int(total_weight)+1}g.gcode

So if the weight was exactly on a gram then it would be 1g too much but I haven't had that happen to me yet and with all the normal weights it in effect rounds up to the nearest gram which was pretty much what I was trying to achieve anyway.

n8bot, 2020-07-01:
> I've been using the following
> {input_filename_base}_{layer_height}mm_{filament_type[0]}_{printer_model}_{print_time}_{int(total_weight)+1}g.gcode

If you add 0.5 instead of 1, you'll get proper traditional rounding. 
E.g.: 
1234.45 = 1234.95 = 1234.
1234.55 = 1235.05 = 1235.

neophyl, 2020-07-01:
I couldn't get that to work when I tried.  It errors on exporting the gcode, although I might have got the structure wrong.  I tried several different ways.  
Besides rounding up to the nearest gram was what I was after with the original issue.  Traditional rounding would drop 12.49g to 12g and I prefer to go up where filament used is concerned.

n8bot, 2020-07-01:
You need to enclose the expression in additional parentheses, like so:

`{input_filename_base}{layer_height}mm{filament_type[0]}{printer_model}{print_time}_{int((total_weight+0.5))}g.gcode`
Results in:
Cube500.04mmPLA2d3h11m_54g.gcode

Where the value of total_weight is 53.7661

Carib3D, 2020-07-01:
Brilliant! Thanks to n8bot & neophyl for the excellent tweaks to the code. My standard hereafter will be:

{input_filename_base}_{nozzle_diameter[0]}n_{layer_height}L_{filament_type[0]}_{print_time}_{int((total_weight+1))}g.gcode

which outputs:

PartName_0.6n_0.3L_PET_3h50m_98g.gcode

Everything I need to know!

n8bot, 2020-07-01:
It would still be nice if print_time could be stated in a different way. Perhaps they could output number of minutes so we can convert into hours, cast to integer, etc.

If you want to avoid rounding down at .49, you can specify a larger number than 0.5 but smaller than 1 to "tune" the rounding to your desire. It seems silly to choose to round 12.05 to 13. Perhaps total_weight+0.75 or 0.8 would be more desireable.

Carib3D, 2020-07-02:
Personally, I like the time format (3h50m) since it suits the timer on my phone, should I want a reminder. I also prefer to round up the weight because I don't fully trust the precision of the slicer, although it's MUCH better than years ago. The main reason I want the weight is to check I have enough filament if I'm not around to monitor and change the spool so a little headroom is good. I may also want to cost a part and the pre-ooze and trimmings should be included in that which likely adds a gram or so.

pavolNovak, 2022-01-30:
Im starting to use Command line to slice files and I have tried this :
{num_objects}x_{input_filename_base}°~{int(total_weight+0.6)}g.gcode
getting this error: 
[image]
I also tried round() but not succesful. I know that my output file format works perfectly in Prusaslicer in GUI , but not in console.

CaviaPorcellus, 2022-06-23:
I have the same issue as pavolNovak. Interestingly, int and round work fine with SLA slicing via the command line (e.g., to round the output of objects_used_material), but FFF slicing gcode output returns errors, namely that the weight isn't a number.