prusa3d/PrusaSlicer #3473: Strange movements generated by wipe tower (spikes ramming the head into the side)

AmyTheCute, 2020-01-06:
### Version
2.1.0

### Operating system type + version
Wiundows 10

### 3D printer brand / version + firmware version (if known)
Custom Toolchanger corexy with klipper

### Behavior
Wipe towers make weird movements ramming the tools it picks to the side. this behavior does not happen without a pruge tower and finishes successfully
[image]

#### Project File (.3MF) where problem occurs
[bunchy.zip](https://github.com/prusa3d/PrusaSlicer/files/4026739/bunchy.zip)

AmyTheCute, 2020-01-06:
[image]
I just noticed the purge tower also makes my head go the home which it can only reach without a tool (it's the tool area) so it just rams the tool there. how can I get rid of this behaviour?

lukasmatena, 2020-01-06:
Likely a duplicate of #3234. Can you retest in 2.2.0-alpha2 and close the issue if you're happy? Thanks.

AmyTheCute, 2020-01-06:
Oh sorry! I didn't see that. it workes thanks :3

closed 2020-01-06 by AmyTheCute