prusa3d/PrusaSlicer #3475: Holes in thin walls

fharding1, 2020-01-06:
### Version
2.2.0-alpha2+linux-x64

### Operating system type + version
Ubuntu 19.04 (Disco Dingo)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3, firmware version is irrelevant

### Behavior

I'm trying to print a pretty complex and small part (2" mecanum wheel) and having some issues with holes in the thin walls it has.

Here's a picture of what the model looks like: https://imgur.com/a/tMtuPOG

Here's what it looks like after being sliced: https://imgur.com/a/gWmboqS

Layer view (the holes on the interior are intentional and in the model, the holes on the exterior are not in the model): https://imgur.com/a/vBwsrBm

As you can see, there are small holes in the walls. No matter what settings I play with, I can't get them to go away. Here's what I've tried:

* Detect thin walls
* Extra perimeters if needed
* Ensure vertical shell thickness
* Perimeter count
* Slice for 0.25mm nozzle
* 0.05mm to 0.15mm layer height
* 0% infill, 100% infill

Is there something I'm missing or is this a bug?

Steps to reproduce:
1. Open PrusaSlicer
2. Add .3MF file available below
3. Select "0.15mm QUALITY" for Print Settings, "Prusament PETG" for Filament, Original Prusa i3 MK3 for printer.
4. Slice, see holes in walls.
5. Try with the settings mentioned above, holes are still present

#### Project File (.3MF) where problem occurs
[MEC Wheel - FRC125 - 2019 - Release Ver.0 -- LEFT_B.zip](https://github.com/prusa3d/PrusaSlicer/files/4027617/MEC.Wheel.-.FRC125.-.2019.-.Release.Ver.0.--.LEFT_B.zip)

neophyl, 2020-01-06:
Your part thickness in that area is about 0.16mm which means even with a 0.25mm nozzle you arent going to get a walls thickness out of it.

fharding1, 2020-01-06:
Thanks for the response. Is there no way to get around that? I know some other slicers have an "expand thin walls" option.

neophyl, 2020-01-06:
I dont think so with PrusaSlicer.  I also gave it a go with Slicer++ (which is built on PrusaSlicer and Slic3r) as that has more options for things like this and I could get it to slice with the holes filled in, however the gaps on the interior of the holes also filled in for 3 of them.  That would indicate that the holes are not perfectly spaced/sized relative to each other and I don't know if the part would work with those filled in.  Basically if you expand the walls for the holes it will effect other parts of the model.

Best fix is to alter the model to have thicker walls where required so that its properly printable.

fharding1, 2020-01-07:
@neophyl got it. Thanks for the help and advice.

closed 2020-01-07 by fharding1