prusa3d/PrusaSlicer #3476: Variable Layer Height, it drastically increases the size of the purge block

rocky00717, 2020-01-06:
### Version
PrusaSlicer-2.2.0-alpha2

### Operating system type + version
Win 10 1909

### 3D printer brand / version + firmware version (if known)
Prusa MK3s MMu2s

### Behavior
When using Variable Layer Height, it drastically increases the size of the purge block.  I do not understand why it would increase the purge block at all. Note: this behavior occurs if "No sparse layers" is selected or not.
To Recreate:
1) Add Model
2) Adjust colors and assign to models
3) Adjust variable layer height.
4) Slice
5) Observe
* _Expected Results_
Purge block should not grow
[image]

* _Actual Results_
[image]

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
[Baby Yoda V2.zip](https://github.com/prusa3d/PrusaSlicer/files/4027689/Baby.Yoda.V2.zip)

neophyl, 2020-01-06:
You have nearly 3 times as many layers when using variable layer height so the bigger purge block is not unexpected as it has to change the filament more often.

rocky00717, 2020-01-06:
I think i figured it out.  A lower layer height means that more length needs to be laid down to equate to the same mm3 purged.

closed 2020-01-06 by rocky00717

lukasmatena, 2020-01-06:
The total amount is of course greater too, because there are more toolchanges (as @neophyl has said).