prusa3d/PrusaSlicer #3477: Optimisation. Take into account the skirt margin to arrange multiple parts

rongith, 2020-01-06:
### Version
2.2.0-alpha2

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
I can't arrange more than 183 parts in that case:
[image]
That is because the arrange feature doesn't take into account the skirt. The parts fit on the plate but not the skirt.

#### Project File (.3MF) where problem occurs
[1074402-r01 - Defaut.zip](https://github.com/prusa3d/PrusaSlicer/files/4027991/1074402-r01.-.Defaut.zip)

dartrax, 2021-11-09:
I have the same problem in 2.4.0 beta 1. Arrange can't be used to fill more than appr. 50-60% of the build plate area of our IDEX printer.
[image]

jschuh, 2021-12-30:
I kept hitting this situation while printing out batches of parts, so I wrote #7653 for my local build. It has its own tradeoff (as described in the PR), but so far I find it much more useful than the current version.

timur-tabi, 2022-05-08:
Would it be possible to specify a way to limit the auto-arrange arrange area, in the same dialog box where you specify the space between objects?  That would, I could say, "auto-arrange these parts into this rectangle" and the auto-arrange would use that rectangle only instead of the whole print area.

codefaux, 2022-10-03:
Suggestion -

1) `Skirt distance + (loops * extrusion width)` applied as mask to perimeter of build area.
- Assume entire build area in use for "minimum extrusion length" calculating loop count, since it'll be guaranteed this size or smaller.

2) `Build margin` parameter for Arrange tool, allowing a generic user-defined margin to be applied to auto arrange calculation

Either seems easy enough to implement, both fix multiple potential issues.

lukasmatena, 2023-04-02:
Arrange function accounts for brim and skirt since [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena