prusa3d/PrusaSlicer #3478: Repeatable crash with 0.30mm Draft MK3 and Generic Flex

joseph-greiner, 2020-01-07:
### Version
2.1.1+win64

### Operating system type + version
Windows 10 Home

### 3D printer brand / version + firmware version (if known)
Prusa MK3, 3.8.1-2869

### Behavior
PrusaSlicer crashes at the end of generating gcode, just before the legend is displayed. No error, it just closes out.

* _Steps needed to reproduce the problem_
Load problem.STL.
Change Print settings to 0.30mm DRAFT MK3
Change filament to Generic Flex
Click on Slice now.

Changing to a different filament preset works. (Generic PLA/Prusa PLA)
Changing to a different Print setting works. (0.20mm QUALITY MK3 / 0.20MM SPEED MK3)
Changing to a different model works.

[Problem.zip](https://github.com/prusa3d/PrusaSlicer/files/4028935/Problem.zip)

lukasmatena, 2020-01-07:
Thanks. This might be a duplicate of #3386, which should be fixed in PrusaSlicer2.2.0-alpha2. Can you confirm that?

joseph-greiner, 2020-01-07:
I never get a chance to save the gcode. When I slice it at 0.2mm SPEED the gcode file is 825KB in size.

lukasmatena, 2020-01-07:
Does it mean you actually tried slicing that in PrusaSlicer2.2.0? Because I did and it didn't crash.

lukasmatena, 2020-03-01:
No response, I couldn't reproduce in current release and all similar issues (#3386, #3747, #2988) are fixed. Closing.

closed 2020-03-01 by lukasmatena