prusa3d/PrusaSlicer #3479: Duplicate mouse_device section in ini - app crash

foreachthing, 2020-01-07:
### Version
2.2.0-alpha2

### Operating system type + version
Win 10 x64 1809

### Behavior
* 3DConnexion settings not restored (set to default values) after restart of PS
* Duplicate ini settings for 
```
[mouse_device:3Dconnexion /SpacePilot PRO 3Dmouse ]
```
after changing settings (CTRL+M) and therefore app-crash on next start:
`Unhandled unknown exception; terminating app`
followed by:
`Internal Error; duplicate section name (line 29)`

* settings will be rewritten after closing. So, need to edit ini file before _every_ start of PS.

### Workaround
Edit ini file, remove duplicate `[mouse_device]` section.

### NOTE
These settings should not be necessary, since 3DConnexion has its own speed settings etc. - use those, please.

enricoturri1966, 2020-01-07:
@foreachthing
I am unable to replicate the duplication of data into the ini file. Could you please add more info in how you get it ?

foreachthing, 2020-01-07:
Well, no more than this:

1. INITIAL start Prusa Slicer
2. CTRL+M for changing the default values (is not needed; just to test if values are read at start)
3. Close PS
4. Start PS
5. Close PS
6. Start PS - Crash.
7. Clean ini
8. `goto 4`

[image]
[PrusaSlicer.ini.txt](https://github.com/prusa3d/PrusaSlicer/files/4030835/PrusaSlicer.ini.txt)

Once PS has been started, I need to clean up the ini file. After closing, PS writes duplicate entries again.... 

If you need more info, please let me know.

foreachthing, 2020-01-07:
After looking at the ini bit more closely .... I can see an unmotivated space:

```
[mouse_device:3Dconnexion /SpacePilot PRO 3Dmouse]
rotation_deadzone = 0.100000
rotation_speed = 2.000000
translation_deadzone = 0.100000
translation_speed = 2.000000

[mouse_device:3Dconnexion /SpacePilot PRO 3Dmouse ]
rotation_deadzone = 0.100000
rotation_speed = 1.500000
translation_deadzone = 0.100000
translation_speed = 1.700000
```

enricoturri1966, 2020-01-08:
My commit automatically closed the issue. This was not intentional ;)

foreachthing, 2020-01-08:
I close here.
I need to wait for the next release to test it anyway ...

closed 2020-01-08 by foreachthing

foreachthing, 2020-01-09:
Just to make one thing abundantly clear: I **LOVE** that I can use my SpacePilot with PS!!!!
Big THANKS to the devs!

commit 7a8251b6 (2020-01-08): Attempt to fix #3479