prusa3d/PrusaSlicer #3481: Redundant setting

espr14, 2020-01-07:
v2.2.0 alpha2 win64

These settings are redundant. Wipe while retracting could be always on so there's no reason to show it. Amount of wipe can be changed by the percentage:

100% = no wipe
0% = full wipe

[image]

BeatSlayer, 2020-01-07:
I believe you didn't understood what those functions are.
Wipe it's a toolhead movement to "break" the plastic string coming out of the nozzle.
The other option is how much of the retraction should be done before initiating the wipe move.

espr14, 2020-01-07:
It does not matter what it says but what it does. Take a model and generate g codes with these combinations:

1. ON, 100%
2. OFF, arbitrary percentage

You'll see that these two g codes are identical. There's no wiping if you set 100%. Therefore, wiping can be controlled by the amount of retraction before wipe. Since this is advanced setting, it does not need to be perfectly clear setting.

BeatSlayer, 2020-01-07:
If that's the case it's a bug and not a redundant setting.

espr14, 2020-01-07:
It's not a bug, it's a feature. It works as it is supposed.

BeatSlayer, 2020-01-07:
Sorry, you're totally right.

bubnikv, 2020-01-09:
>  It's not a bug, it's a feature. It works as it is supposed.

Then please close the issue. Thank you.

espr14, 2020-01-09:
This issue is about removing an item in the settings not about algorithm.

bubnikv, 2020-01-10:
> This issue is about removing an item

Why? The item has its purpose.

closed 2020-01-10 by bubnikv