prusa3d/PrusaSlicer #3482: v2.2.0 alpha2 Unhandeled exception - extrusion width

espr14, 2020-01-07:
I was trying to set extrusion width of first layer and perimeters to 1.2 mm (with 1 mm nozzle). Selecting NO caused no change. Selecting YES caused unhandled exception:

[image]
[image]
[image]

FidelCapo, 2020-01-10:
Thank you for report. You are right, this is issue.
Created JIRA-SPE-1069.

bubnikv, 2020-02-18:
The unhandled exception is fixed with 43f2171. I believe this is an old issue.

FidelCapo, 2020-02-20:
This issue should be fixed in next release.
Closing.

closed 2020-02-20 by FidelCapo