prusa3d/PrusaSlicer #3484: Limit size of calibration

espr14, 2020-01-07:
v2.2.0 win64

G29 command allows to define area which should be calibrated. Slicer can take first layer including brims, skirts and purge tower and make a bounding box. Then put it in the G29 command.

With limited size of calibration area, the calibration (mesh bed leveling) can be faster and/or more precise.

ondrasy, 2020-06-20:
Maybe it would be sufficient to make some new placeholders for the custom g-code that would provide dimensions and positions of to-be-printed objects. The boundaries for G29 command would then be easy to compute.

rtyr, 2021-02-05:
Implemented, see https://github.com/prusa3d/PrusaSlicer/issues/5181#issuecomment-733140455. Closing.

closed 2021-02-05 by rtyr