prusa3d/PrusaSlicer #3487: Smal texts in UI on high DPI screen

noradtux, 2020-01-08:
### Version
2.2.0-alpha2-33-gca950d1a0
Compiled using prusa-slicer-git from AUR:
```
  mkdir -p build

  cd build
  cmake ..\
    -DSLIC3R_STATIC=0 \
    -DSLIC3R_FHS=1 \
    -DSLIC3R_WX_STABLE=1 \
    -DSLIC3R_GTK=3 \
    -DSLIC3R_PCH=0 \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_INSTALL_LIBDIR=lib

  make
```

### Operating system type + version
Arch Linux

### Behavior
Some texts are very small in 2.2 alpha2. On 2.1 all looked normal. This is a High-DPI setup on KDE.
[image]

YuSanka, 2020-01-08:
Thanks for your report. Unfortunately, we couldn't reproduce this bug on Ubuntu (no one of our developers uses ArchLinux currently). 
Since you've build PS yourself already, could I ask you for a favor to build it again with a one line change? I think that this bug can be fixed by adding of code line:
 `m_treectrl->SetFont(wxGetApp().normal_font()); `
into Tab.cpp: 216
It will help us to resolve the issue much quicker than if we built it on Arch ourselves. Thanks!

noradtux, 2020-01-08:
Doesn't seem to help.

noradtux, 2020-01-25:
Ok, in https://github.com/prusa3d/PrusaSlicer/issues/3571 it was suggested that GTK3-build are not tested much. So I gave it a shot. Using GTK2 fonts are displayed correctly. So this seems to be a GTK3 issue.

Salamandar, 2020-04-07:
I guess this is related to https://github.com/prusa3d/PrusaSlicer/issues/3245

noradtux, 2020-04-10:
This is most definitely a bug in wxWidgets. By now I've seen it in other wxWidgets-apps aswell (KiCAD).

lukasmatena, 2020-12-18:
Can you please retest this with [2.3.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0-rc1)? Thanks.

simon-budig, 2021-02-17:
I can confirm that the problem still is present in version 2.3.0.

This definitely is related to the use of wxwidgets. Using the gtk inspector the relevant texts always are hidden inside a "wxpizza" widget and there is no way to access the text properties.

You can open the GTK inspector by running

$ GTK_DEBUG=interactive prusa-slicer

A hint that something is wrong is when you change the "font scale" property in the "Visual" tab of the inspector. You can see that the regular GtkLabels react to that, but the text inside the wxpizza widgets does not.

noradtux, 2021-02-18:
As far as I can tell this is an issue in wxgtk3. KiCAD for example shows 
the same issue.

/Malte

noradtux, 2021-02-18:
Addendum: I just recompiled Prusa-Slicer against wxgtk 3.1.4 (-dev). 
Text scaling works as expected there.

/Malte

simon-budig, 2021-02-18:
I can confirm that my version of wxgtk is older: I am running debian unstable which has libwxgtk-3.0 3.0.5.1

lukasmatena, 2023-07-10:
I believe all these issues should be gone now when we transitioned to wxWidgets 3.2. Closing.

closed 2023-07-10 by lukasmatena