prusa3d/PrusaSlicer #3492: [2.2.0-alpha2] Export toolpath as .obj looses outer shell ( only infill in .obj )

schiffer-soft, 2020-01-08:
v2.2.0 alpha2 ( and below, but not tested ) win64

I think I found an issue in the "export toolpath als obj" funktion.
If you load the Europaturm.stl into the slicer with default settings ( i3mk3 MMU2 0.15mm ) and slice it. Works flawlessly for me.

[image]

Than export via file-> export toolpath as OBJ
If u now open the obj file with win10 3D-Viewer or Fusion 360 it only shows the infill.
The outer shell its gone... why? ( I tested it on multible PCs all win64 )

[image]

[Europaturm.zip](https://github.com/prusa3d/PrusaSlicer/files/4035252/Europaturm.zip)

MFG

Chris

enricoturri1966, 2020-01-09:
Working on it -> https://dev.prusa3d.com/browse/SPE-1068

lukasmatena, 2020-01-31:
This should be fixed in master by 63f8fda for the next release (2.2.0-alpha3). Closing.

closed 2020-01-31 by lukasmatena