prusa3d/PrusaSlicer #3495: Modificator problem

PaaXus, 2020-01-08:
### Version
2.2.0 alpha2

I used modificator with infill and perimeters... That modificator cut perimeters on his border.
[image]

[image]

When i deleted modificator all looks good.

[image]

espr14, 2020-01-08:
If you change perimeters it will cut the model. It's easier to program it this way. There's a workaround: decrease size of the modifier so it does not go out of the model. In other words, it has to be inside and touching walls from inside. Then the external perimeters is not crossed thus it's printed continuously. This will be harder in more complex shapes (more modifiers needed) but it's doable.

PaaXus, 2020-01-09:
OK, but it's little bit laborious. Thank you for your advice.

bubnikv, 2021-06-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3370

closed 2021-06-03 by bubnikv