prusa3d/PrusaSlicer #3496: Slicer crashes

fitzroy99, 2020-01-08:
### Version
2.1.1 + win64_and 2.2.0 alpha2

Win7 Home Edition

Practicing while waiting for my Mini
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

* _Describe the problem_Program crashes every time I push 'slice now', even with .stls that have worked fine on Cura 4.3.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2020-01-09:
I'm afraid we will need more info than "slicer crashes". A 3MF with the failing scene would be a good start. Like you yourself posted:
> Upload a PrusaSlicer Project File (.3MF) (Plater -> Export plate as 3MF for Slic3r PE 1.41.2 and older, File -> Save / Save Project for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)

fitzroy99, 2020-01-10:
I get error message "unable to enumerate HID device".

bubnikv, 2020-03-15:
I suppose this is duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3539

closed 2020-03-15 by bubnikv