prusa3d/PrusaSlicer #3497: Disabling Retraction in Filament Settings doesn't work with Firmware Retraction

thedevleon, 2020-01-08:
### Version
PrusaSlicer-2.2.0-alpha2+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Heavily Modified Ender 3, Duet WiFi, RepRapFirmware for Duet 2 WiFi/Ethernet 2.05

### Behavior
I have firmware retraction enabled, and even though I disabled retraction in the filament settings by checking the override and setting the length to 0, G10 and G11 retraction commands are still generated in the final .gcode file.

[image]

[image]

#### Project File (.3MF) where problem occurs
[xyzCalibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/4037211/xyzCalibration_cube.zip)

lukasmatena, 2020-01-09:
I looked up the code and the behaviour seems to have been added on purpose more than 5 years ago by alranel when he was fixing https://github.com/slic3r/Slic3r/issues/2359. Someone didn't want zero length retraction to disable firmware retraction :-) Although that was before Filament Overrides were added.

Honestly, I'm not sure what to do about it and what the correct behaviour is. Changing it back is not backward compatible and can potentially break existing printer profiles.

thedevleon, 2020-01-09:
I see. Seems like the issue there though was a bit different though, in the regard that the retraction field (in the printer settings) simply became disabled and couldn't be changed when firmware retraction was enabled, leading to incorrect behaviour if it was set to 0 prior. 
I agree that this value should be ignored when firmware retraction is enabled.

For the Filament Overrides, however, I think the value of 0 should not be ignored. Can you elaborate on how this would not be backward compatible and could break existing profiles? If this setting was enabled and 0 prior to the change, someone wanted to disable retraction anyways. If it was enabled and set to another value, it would have been ignored anyway (as it is now), since firmware retraction GCodes don't handle length.

lukasmatena, 2020-01-09:
I didn't think about it too much, but I think that the combination retraction length=0 and firmware retraction enabled does retract now. Just reverting that change will break any profile that is set that way - and no, I don't know how many there are.

You are correct that now when we have Filament Overrides it makes sense that if you set zero in there, retractions should be ignored regardless if they're firmware or not. I'll add this to my todo list.

ChrisVahi, 2020-04-25:
try setting your retraction length to something so low (eg 0.0001mm) that there is no effective retraction. This is what I do to use Z hop without retraction with carbon fiber filament.

guytas, 2021-01-08:
the 0.0001mm doesn't stop the retraction. :(

lukeski14, 2021-01-12:
Can confirm this is still an issue. Just ran into it while printing TPU.

guytas, 2021-01-12:
sorry... no.  I had loaded an INI file for the printer configuration and I think it wasn't compatible. I had all sort of weird issues until I restarted the printer.  Now it's OK.  Entering 0 seems to stop retracting completly.

Thanks

haleypearse, 2022-04-01:
Confirmed this has been fixed in Superslicer, but I'm still seeing the retracts on the preview and in gcode with Prusaslicer.

Fuckingnameless, 2023-08-07:
helloooooooooo

closed 2026-09-29 by prusaslicer-bot[bot]