prusa3d/PrusaSlicer #3498: Speed up straight lines

espr14, 2020-01-08:
v2.2.0 alpha2 win64

Adhesion of thread is fine at straight lines but cause trouble at corners. So longer straight lines could be printed even faster.

There could be options that define max.speed at straight lines and length of safe zone. Each (long) straight line (1 G1 instruction) would be split into 3 parts (3 G1 instructions):

1. safe zone
2. max. speed zone
3. safe zone

Safe zones are a must to ensure proper adhesion of layers. Otherwise, the thread would detach and take a shortcut at the corner. There are no side forces at the middle of the straight line so there's only a visual aspect. It should not be used for external perimeters and bridges/overhangs. Still, in case of draft external perimeters are still doable. Rectangular wipe tower would benefit as well.

Acceleration could be retained. In case of bad tuning of linear advance, special acceleration could be used which should be way more slower than the standard one. This would require extra 2 instructions (changing acc. there and back). The lower the acc. the shorter safe zone can be.

johnbiggs, 2020-04-23:
Isn’t this what first layer acceleration does?

espr14, 2020-04-23:
It is not. Acceleration speeds up from the beginning. This proposal accelerates to start speed and after a while it accelerates to speed which is higher than usual max speed.

lukasmatena, 2024-02-18:
Now when we have printers with IS and quite some experience with what is needed to make them print fast, I believe this can be closed.

closed 2024-02-18 by lukasmatena