prusa3d/PrusaSlicer #3499: Alpha 2.2 MMU change .2 to .3 heights increases purge block X4

midnightsmith89, 2020-01-09:
### Version
Alpha 2.2

### Operating system type + version

windows 10 pro

### 3D printer brand / version + firmware version (if known)
Prusa MK3s with MMU2s with October firmware

### Behavior
When slicing a model for MMU print in standard prusa slicer, current release, changing from 0.2 to 0.3 layer heights is ok.  When using Alpha 2.2, it increases the purge block approx 4X the size.

load multi material object into Alpha 2.2, slice at 0.2, block is normal.  Re slice at 0.3, block is huge!
 
* _Expected Results_
Expect to maybe increase a tiny bit in size, not use most of the build plate 
* _Actual Results_
purge block is massive and cannot be made smaller
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
See pics at twitter https://twitter.com/midnightsmith/status/1215059726674755584
_Is this a new feature request?_
no
#### Project File (.3MF) where problem occurs
not limited to one specific file

lukasmatena, 2020-01-09:
Would you bother to post a 3MF, please?

midnightsmith89, 2020-01-22:
I have reinstalled the alpha version and it seemed to go away, I could not figure out how to upload a 3MF file to here.  I'll consider this a bug with installation and call it closed.

closed 2020-01-22 by midnightsmith89